SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #104: Micro Mages video glitches

bgulotta, 2021-03-05:
Within a few minutes of playing, I am seeing very bad video and audio issues with the Micro Mages cartridge on the newest version of the jailbreak firmware.

tjanas, 2021-03-05:
Can you share a photo and/or video clip?

bgulotta, 2021-03-05:
Sure let me get one uploaded.

bgulotta, 2021-03-05:
https://youtu.be/Nu1ChD56Cw4

tjanas, 2021-03-05:
Thank you for the video. Have you tried a different cart voltage setting? What about if you set the Noir to directly boot the cartridge? Either way, it’s a Noir bug for sure.

bgulotta, 2021-03-05:
I tried a few other carts and they worked just fine. I don’t recall this issue with the previous versions of the jailbreak. I can try and restore to a previous version and or restore the stock firmware to try and identify which version caused the issue.

dot-bob, 2021-03-09:
I have been through the first and second levels using the micro mages physical cart and the rom with my N8 Pro Fami using the 6.6 jailbreak without any glitching.  The problem may be with your cart.

bgulotta, 2021-03-12:
I highly doubt there is anything wrong with the cart, it is practically brand new. Also, the game ran perfectly fine on previous versions of the jailbreak and the stock firmware. I'm going to try and reflash 6.6 and see if it was an issue with the flashing process.

bgulotta, 2021-03-12:
I verified it is definitely the 6.6 jailbreak that the issues start. I reflashed 6.5 and the issue went away. Reflashed 6.6 and the issue comes back. I'll just stay on 6.5 for now. I do have 2 wireless n30s attached to the noir. I'm not sure if that has anything to do with it or not.

pezillionaire, 2021-03-12:
@bgulotta and @tjanas I have a Micro Mages cart and can boot it up this afternoon/evening to verify. 

Related to this: I also have Project Blue cart and it very much as graphical issues on the latest firmware `JB6.6` with the default `3.6V` voltage, switching to `5V` initially seems to resolve the issue (more playthrough is required to verify) but this has ONLY popped up with the latest fimmware. Playing Project Blue ROM off the SD card has none of the graphical issues.

dot-bob, 2021-03-12:
I am using the controller provided with the noir.  The rom used on the everdrive pro I obtained through my Steam purchase of the PC version of micro mages.  The cart I obtained from a US Private Seller around Dec 2019 selling a used copy for $40.  As far as I can tell it looks legit with box and manual, but still could be a china cart as I haven't cracked it open.  All I have compared it with are images from Morphcat's website.

bgulotta, 2021-03-12:
I’ll try it with a single controller later tonight and get back to you guys. My copy of Micro Mages was purchased directly from Morphcat, so I know it is a good copy.

GR33NDevil, 2021-03-13:
I am also having this issue. Nt Mini Noir on latest JB. Seems to glitch less when cartridge voltage is set to 3.6 but eventually happens.

pezillionaire, 2021-03-13:
Ok I have confimed that having the `3.6v` toggle off/false does cause graphical glitches in Micro Mages.
This causes similar issues in Project Blue as well.

While this does appear to be a "new" bug in `JB6.6` per my experience —  I'm not sure if this is actually unintentional. Per my understanding (via the NT Mini Noir manual) the purpose of this toggle is infact to resolve issues with modern NES carts that run a a higher voltage.

GR33NDevil, 2021-03-13:
Yes but I still get glitches when toggling 3.6v

pezillionaire, 2021-03-13:
@GR33NDevil regardless of toggle the `3.6v` setting the graphical issues still occur? Curious that I don't get that…

Can you describe the graphical issues or where they happen? 

When I toggle the `3.6v` to `true` and re-boot the cart there are no obvious graphical glitches. Conversely, when I toggle the `3.6v` to `false` and re-boot the cart — the graphical glitches are almost immediate upon loading of 1-1 (the splash/sizzle screens appear to have no issues regardless from my experience).

GR33NDevil, 2021-03-14:
I thought the same thing but played through all of the first stage to the
boss and glitched through the wall a small handful of time. Not immediate
but still happens

adam16bit, 2021-05-14:
I have the same problem - but with slightly different glitches, different almost every time.   
It works fine with the regular Analogue 1.2 firmware.

It glitches out with the Jailbreak 6.6 firmware.  (But the ROM runs fine.  Only the cartridge  has problems, and only with the Jailbreak firmware.)  I've had similar problems with two more recent releases - From Below and Project Blue cartridges.

In my case, the voltage changes make no difference whatsoever so far.  But the firmware changes clear up the problems immediately - it's only the Jailbreak that's giving me trouble, and it happens within a few seconds of opening the first stage.

tjanas, 2023-05-04:
@bgulotta @GR33NDevil @pezillionaire please re-test your cartridges with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

GR33NDevil, 2023-05-06:
just tried with the new firmware. Issue persists on level 2 but only on the first screen of side scrolling. Once the second area is reached, the bars stop jumping.

eltees, 2023-05-06:
FWIW I booted up my micro mages cart and played through to world/dungeon 4 with both 3.6v toggled off (default) and on and didn't see the issue either way on the current firmware.

This is with 100% system defaults post-update (haven't gone back and customized anything yet).

Though to be completely fair i never saw the same issue prior to the update either, and broadly my noir has never really had any connectivity issues for nes carts though i haven't used carts super often. I know on the prior firmware I had played through similarly to world 4 or 5 with friends on the cart back in the day before putting JB on.

tjanas, 2023-05-06:
I am going to close this issue. If @bgulotta (OP) still has a video issue that can be confirmed by another user, this can be reopened. @GR33NDevil maybe something else is going on with your system or cart that isn’t related to the firmware?

closed 2023-05-06 by tjanas

GR33NDevil, 2023-05-06:
Ignore my previous comment. Micro Mages worked fine after the new firmware. 

I mistakenly replied thinking this was the issue opened for Vice Project Doom.