SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #105: Devil Man (jap) background graphics filled with zeros, running from SD Card

jengelz, 2021-03-06:
Check the pictures, running JB FW v. 6.6, works fine on Emulation Station with FCeux on Retroarch
[image]
[image]

GreatHierophant, 2021-03-07:
Works fine for me.  Make sure the mapper is set in the NES header to 154.  This game is the only game which uses mapper 154.

jengelz, 2021-03-07:
Ok, so i have to adapt the header inside the rom? Because i don't see any option on the Nt Mini Noir to change this, when you run the game off of SD Card. I searched the entire menu.  Maybe with CopyNES, i'm not sure...I'm not so familiar with Headers. If you have a link or something on how to do it, would be great. Many thanks

tjanas, 2021-03-07:
@jengelz please see https://github.com/Kitrinx/NES_Header_Repair

jengelz, 2021-03-07:
I ran the header script. The ROM is now adapted from header 88 to 154. But now the ROM has dissapeared from my ROM list inside my NES folder on the SD Card. The Mini cannot read or recognize it  anymore...
[image]

GreatHierophant, 2021-03-07:
You'll have to find where the script moved the ROM.

jengelz, 2021-03-07:
You mean on my PC or on my NT Mini? The script just updated the Header of the ROM in the same folder. I did all that on my PC. I then deleted the rom on my SD Card. Then copied the updated ROM from PC to SD Card. At least that's how i understand it. I can't read or understand script (or very minor understanding) Thans for helping

closed 2021-03-09 by tjanas