SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #107: Video Glitches - Dragon Crystal - SMS Core

Ulisses-Medeiros, 2021-03-13:
I have video glitches while playing Dragon Crystal(Europe) - CRC32:9549FCE4, the errors appear and are displayed while playing, but when you press the button to open the menu in the game, and you return to the game, the glitchers go back and forth after playing for a while, the time is random, it is not possible to measure how much in how much happens.
[image]

I played the same game using JB 6.5 and 6.6 and have same problem on NT Mini Noir, while on mega sg with JB 7.8 and there were no errors, so it's probably JB on NT Mini Noir.

A doubt, are the core of Genesis and Master the same ? I'm also having problems with the core of Genesis, games freeze suddenly, and I still have access to the system menu and I can change rom, plus other games like Sonic 2 (World - Rev), elemental master (USA and JAP), also freeze.(a will open a new issue if required)

tjanas, 2023-06-26:
@Ulisses-Medeiros Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

closed 2023-06-26 by tjanas

Ulisses-Medeiros, 2023-06-26:
As soon as possible I test and publish