SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #108: Freezing and Graphic Glitches from Cartridge - Shadow of the Ninja

Arcnarenth, 2021-03-25:
Shadow of the Ninja displays graphical glitches and often freezes during the opening cutscene when played from cartridge on JB 6.6.  Controller input also appears frozen and the game is unable to be started.  

The same cartridge was tested on a top loading NES and it works as intended.  A ROM loaded through the NES core of the jailbreak appears to behave properly as well.

Demonstration video:
https://youtu.be/FjxW6jvVHFE

whitezombie2000, 2022-02-14:
I cannot reproduce this issue with my copy Shadow of the Ninja using a real cartridge. It works with 6.6 JB and official 1.2 firmware. I successfully dumped the ROM with CopyNES using Generic MMC3 TxROM, the dumped ROM works on 6.6 JB without issue, and work on an emulator. It should be noted that the back of my cartridge says REV A.

tjanas, 2022-02-14:
@Arcnarenth does this issue still occur for you? Does it happen every time? What settings are you using for Noir? Fully buffered, zero delay, etc?

whitezombie2000, 2022-02-14:
I just have the default settings after flashing the firmware when I do my testing.

Arcnarenth, 2023-05-05:
@tjanas Issue is resolved with firmware 1.3.

closed 2023-05-05 by tjanas