SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #109: Rollie Homebrew cart (GTROM) glitches, freezes

dlbuller, 2021-03-26:
I can't seem to get Rollie, a homebrew cart, to work correctly.  Start screen it glitches a little, sprites disappear, then freezes.  I emailed the creator and he told me its a GTROM.  

https://user-images.githubusercontent.com/25686427/112639847-1bfeaf80-8e17-11eb-9a73-a2ebb073e4b3.MOV

tjanas, 2021-03-26:
Is your Noir set to run in 3.3V or 5V? Do you have dejitter enabled or disabled?

dlbuller, 2021-03-26:
> Is your Noir set to run in 3.3V or 5V? Do you have dejitter enabled or disabled?

Good question. I was going to start playing with cart voltage today sometime. 

Currently, everything is set to whatever the default is for the firmware.  Unsure of what the default settings are but I can follow up later.

tjanas, 2021-03-26:
What firmware version are you running?

dlbuller, 2021-03-26:
JB6.6

dlbuller, 2021-03-26:
I’m seeing the same issues.  Selected 3.6 cart, unselected it, tried combinations of dejitter. Not sure what to try next. Thanks!

tjanas, 2023-05-04:
@dlbuller please re-test your cartridge with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

dlbuller, 2023-05-05:
Unfortunately this happened 2 years ago. I no longer have the hardware.

Assigned #109 to @dlbuller.