SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #110: Using Battletoads with the correct header from Smokemonster pack, the game freezes on the 2nd stage

wiselinkpro, 2021-03-30:
Battletoads freezes on the 2nd stage on the Analogue NT Mini Noir jailbreak, using the rom file.  However I spoke with someone with an original NT Mini (non-noir) and he claims it works fine on his.  Maybe this is due to the latest jailbreak changes with the game who knows.  If you use the game genie code to skip level 2, the game plays fine.

tjanas, 2021-03-30:
Can you confirm the NES header bytes are the following:
`4E 45 53 1A 10 00 70 08 20 00 00 07 00 00 00 01`

Dejitter: disabled or enabled?

wiselinkpro, 2021-03-31:
Here's what it says in my hex editor:
4e 45 53 1a 10 00 70 00 00 00 00 00 00 00 00 00

tjanas, 2021-03-31:
Here is the checksum for the file in the [Noir SMDB](https://raw.githubusercontent.com/frederic-mahe/Hardware-Target-Game-Database/master/EverDrive%20Pack%20SMDBs/Analogue%20Nt%20Mini%20Noir%20Jailbreak%20SMDB.txt):
**Filename**: `NES/1 US - A-F/Battletoads (USA).nes`
**SHA256**: `2cc2a9afae9b9e875f0fd2f36affb615d81ce4301dbf530336fc23add999e63d`
**SHA-1**: `b5e4af9df63361498ed54a4e71bfe01e8d1a1481`
**MD5**: `1008f948966f7acc626bb5bf70c5089a`
**CRC32**: `cfff05a1`

wiselinkpro, 2021-03-31:
thanks now I have to figure out where to get the correct file because all the ones I've been downloading are exactly the same as what I already have

tjanas, 2021-03-31:
Have you tried using the NES 2.0 XML Database and the NES Header Repair Tool Python Script to validate and fix your rom? See the README.

tjanas, 2021-03-31:
If you remove the 16 byte header and checksum Battletoads, it should match [no-intro](https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=0229):

Size: 262144
CRC32: 279710DC
MD5: EEFAC97AC3B85048F3685E1623210EFB
SHA-1: D85C9FF489672534FBF61A15F8FA56FFF489A34B

NES 2.0 XML Database:
```
	<game><!-- Licensed North America\Battletoads.nes -->
		<prgrom size="262144" crc32="279710DC" sha1="D85C9FF489672534FBF61A15F8FA56FFF489A34B" sum16="18CC" />
		<rom size="262144" crc32="279710DC" sha1="D85C9FF489672534FBF61A15F8FA56FFF489A34B" />
		<chrram size="8192" />
		<pcb mapper="7" submapper="2" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```

wiselinkpro, 2021-03-31:
In that case my ROM must be bad, I'm getting an entirely different CRC32 than that after removing the first 16 bytes.  Which rom set is your battletoads from? Apparently mine originated in goodnes.

wiselinkpro, 2021-03-31:
just deleted my comment because now I have a matching CRC to your last post.  I'll try it out.

wiselinkpro, 2021-03-31:
yup it froze but now I have the correct header

tjanas, 2021-03-31:
Is the Dejitter enabled or disabled?

wiselinkpro, 2021-03-31:
It was disabled.  I enabled it and got a little further then it froze again.

wiselinkpro, 2021-03-31:
I'm guessing this game only works on the original Analogue NT Mini jailbreak, not the noir.

Ulisses-Medeiros, 2021-04-02:
when you speak in stage 2, is it the one who goes down the rope? If yes, I got to the part of the bike and without any freezing, if possible specify the version of the jailbreak and the quality of the micro sd used

GreatHierophant, 2021-04-12:
Battletoads (USA) works OK in jailbreak v6.5.  If you wish to play it in v6.6, you can use the warp at the beginning of stage 1 to bypass stage 2 completely and get to stage 3.  The game does not appear to crash except for stage 2, and it does not crash in Battletoads (Japan) or Battletoads (Europe) on stage 2.

wiselinkpro, 2021-04-12:
Ok thanks, I wonder if they will fix it in v6.7 or whatever version they come out with next.  I'll have to check and see if I have the europe version on there and run that instead.

wiselinkpro, 2021-04-12:
Europe version is worse. The only one that plays good is Japan but half the text is in Japanese.

tjanas, 2021-04-12:
Hopefully the next Noir firmware squashes all of these outstanding issues without regressions. Maybe having smaller but more frequent firmware updates would be a good thing.

wiselinkpro, 2021-12-27:
Wow I posted about this clear back in March and nothing has been done since February.  Tsk... oh well maybe by 2025 there will be an update.

wiselinkpro, 2022-01-10:
Nevermind on this.  Ti released a rom hack which fixes all the bugs in this game.  I'm playing it right now it works 100% flawless on current jailbreak fw.

tjanas, 2022-01-10:
Reopening; just because a rom hack exists that works around the issue, it doesn’t fix the issue with the actual cartridge.

wiselinkpro, 2022-01-10:
Good enough for me I closed it.  After all I don't even own one cartridge, not even the everdrive.  Been almost a year since an update, doesn't look like anything is happening.  It's fine.  Everything I want works.  Coleco was a damn good bonus, too.  Sold all my Coleco systems and games.

whitezombie2000, 2022-02-16:
@tjanas Was the bug reported to Analogue Support or do the devs hang out in that channel? I was going to test this on a real cart to see what firmware it happened on. I'm assuming by the date it was 1.2 official or 6.6 JB since 1.2 was released Feb 22, 2021.  Was also curious if the bug is present in the older 1.0 or 1.1 firmware on a real cart.

tjanas, 2022-02-16:
I doubt that Analogue devs monitor the discord channel, it wouldn’t hurt for someone with the cart to contact Analogue Supporr directly.

whitezombie2000, 2022-02-16:
I can confirm that this crash happens using a real cartridge with official 1.2 firmware and doesn't crash with 1.1 firmware. As soon as you begin Level 2 Wookie Hole, the crash will occur, it maybe instant, or it may take about 10 seconds. It did the same thing on 6.6 JB firmware after I dumped the ROM with CopyNES using AxROM. I ran the ROM and it crashed at the same point. I will email Analogue Support with a description of this issue.

[image]

[image]

[image]

wiselinkpro, 2022-02-16:
Just use the ROM in smokemonster's hacks folder.  Someone already fixed all the glaring flaws in the game which are what cause it to crash.  I don't have any issues whatsoever with my Analogue NT Mini Noir any more since I found that folder on my microsd card lol.  It has been a year since they released a new fw, I'm not holding my breath.  It's fine though.  This system is pretty awesome.  I just wish it had TG16 support, then it would be perfect.

tjanas, 2022-02-16:
It is unacceptable for a real licensed cart to crash on the Noir. The system is still far from perfect; it has regressions vs. the original Nt Mini from 2017 which could work correctly with an FDS RAM Adapter in zero-delay mode (it does not work with Noir in zero-delay mode).

tjanas, 2023-05-04:
> I can confirm that this crash happens using a real cartridge with official 1.2 firmware and doesn't crash with 1.1 firmware. 

@whitezombie2000 please re-test with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

It should also be re-tested with the future JB firmware.

wiselinkpro, 2023-05-04:
Cool thanks for the heads up. I wonder when jailbreak will be released and if it will include anything new.

whitezombie2000, 2023-05-05:
I can confirm this no longer an issue with Battletoads. I was able to get though Level 2 Wookie Hole with no freezes on v1.3 firmware.

closed 2023-05-05 by tjanas