SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #111: Gradius II (J) dumps incorrectly using "Konami VRC4.BIN"

Bohman-Timothy, 2021-04-01:
According to [https://nesdir.github.io/](https://nesdir.github.io/), Gradius II uses KONAMI-VRC-4, mapper 25. I used the corresponding "Konami VRC4.BIN" CopyNES plugin, and as you can see in the screenshots taken from the dumped ROM, the graphics are scrambled but bear some resemblance to at least the outline of the actual shapes.

[image] Opening graphic of the player's spaceship, which appears to use the correct color scheme within the boundaries of the ship's form, thus resembling the actual ship's graphic.

[image] For comparison, here is the normal version of the opening graphic.

[image] Title screen

[image] Weapon upgrades selection screen

My dumped ROM of the game runs (I haven't tried getting past the first section because the messed up visuals make it extra difficult, so I don't know how much of the game can be played) and the sound effects and music work in the portion of the game I have been able to test. The correct ROM file occupies 256 KB; the dump I get of the game occupies 384 KB. I dumped using the firmware "ntmv2_firmware_verJB6.2.bin" and later with "ntmv2_firmware_verJB6.6" and got the exact same file as output. An 8 KB save file is also created, even though Gradius II doesn't use a battery save.

Bohman-Timothy, 2021-04-10:
Just thought I'd update to say that I managed to play through to the end of the game (with rewinding). So the dump does contain the game, just with mostly the wrong graphics tiles being loaded. The sound and music all seemed correct, as far as I could tell. I included some more screenshots below.

Anyway, this is the only official licensed game of mine released during the lifespan of the NES that doesn't dump correctly, aside from a few overdumps that were easily fixed by using a hex editor to truncate the file. I'd really appreciate it if this could be fixed.

[image] This is the escape sequence, where your ship flies through some kind of rectangular tunnel.

I found where many of the text letters and numbers are used.

[image] This is a solar flare which has its tiles replaced by letters and numbers.

There are also some graphics that are correct or partially correct.

[image] Most of the bird looks correct when its wings are in this position.

[image] The eyeball and the skull look mostly correct, as well as most of the border.

tjanas, 2021-04-11:
Can you confirm the dumped rom does not match no-intro after you strip its iNES header?

Bohman-Timothy, 2021-04-11:
This is the top-level info I get from the [ROM Hasher](https://www.romhacking.net/utilities/1002/) utility:
```
ROM not found in Database: No-Intro
File SHA-1: 4B9423E2A392B8AE9608533402738D3D177BA42A
ROM SHA-1: 795E62AE6A177AEEEDC4383E7BA3F636EC91858B
```
After removing the header by using a hex editor, I get the same ROM hash, which is to be expected:
```
ROM not found in Database: No-Intro
ROM/File SHA-1: 795E62AE6A177AEEEDC4383E7BA3F636EC91858B
```
Below is the rest of the info given by the ROM Hasher utility for the headered ROM.

**General:**
```
Platform           Nintendo Entertainment System
ROM format         INES
External Header    Yes
File Size          393232 (60010)
ROM Size           393216 (60000)
ROM CRC32          BCBE5AAE
ROM SHA-1          795E62AE6A177AEEEDC4383E7BA3F636EC91858B
```
**All Hashes:**
```
File MD5      E2B65EC0949ED4A80CE436950848FB6B
File SHA-1    4B9423E2A392B8AE9608533402738D3D177BA42A
File CRC32    94D58379
ROM MD5       43CBA2A62DEAB5C252D985778594C237
ROM SHA-1     795E62AE6A177AEEEDC4383E7BA3F636EC91858B
ROM CRC32     BCBE5AAE
```
**Header:**
```
CHR Banks          32
PRG Banks          8
Battery backed     No
Mapper             23 ($17) - VRC2_B
Mirroring          Vertical
Region             NTSC
Trainer present    No
Bus conflicts      No
VS Unisystem       No
Playchoice 10      No
iNES 2.0           No
```
And here's a screenshot to confirm the utility and its output (using the database files included with the utility):
[image]

equaldracos, 2022-04-15:
I have Crisis Force, it uses 23.2  - VRC4e mapper and it has a similar problem, I dumped using ntmv2_firmware_verJB6.6. The rom is playable but the graphics are messed up.
[image]

tjanas, 2022-04-15:
> I have Crisis Force, it uses 23.2 - VRC4e mapper and it has a similar problem, I dumped using ntmv2_firmware_verJB6.6. The rom is playable but the graphics are messed up.
> [image]

If you strip the header from the dumped ROM, does the checksum differ from the no-intro checksum? https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=0455

Cheesemeister3k, 2022-11-13:
I tested my Famicom games with Konami mappers and got similar results.

KONAMI-VRC-1 (75) using multiple Konami VRC*.bin files:
Ganbare Goemon! Karakuri Douchuu
Tetsuwan Atom
PRG OK, CHR NG

KONAMI-VRC-2a (22) using VRC4:
Ganbare Pennant Race!
TwinBee 3: Poko Poko Dai Maou
PRG OK, CHR NG

KONAMI-VRC-2b (23) using VRC4:
Contra
Dragon Scroll: Yomigaerishi Maryuu
Getsu Fuuma Den
Jarinko Chie: Bakudan Musume no Shiawase Sagashi
Konami Wai Wai World
PRG OK, CHR OK

KONAMI-VRC-2c (25) using VRC4:
Ganbare Goemon Gaiden: Keita Ougon Kiseru (Rev 1)
PRG OK, CHR NG

KONAMI-VRC-3 (73) using VRC3:
Salamander
PRG OK, CHR OK

KONAMI-VRC-4b (25) using VRC4:
Gradius II
PRG OK, CHR NG

KONAMI-VRC-4c (21) using VRC4:
Ganbare Goemon Gaiden 2: Tenka no Zaihou
PRG OK, CHR NG

KONAMI-VRC-4d (25) using VRC4:
Teenage Mutant Ninja Turtles
PRG OK, CHR NG

KONAMI-VRC-4e (23) using VRC4:
Akumajou Special: Boku Dracula-kun
Crisis Force
Parodius Da!
Tiny Toon Adventures
PRG OK, CHR NG

KONAMI-VRC-7 (85) using VRC7:
Lagrange Point
PRG OK, CHR OK
Tiny Toon Adventures 2
PRG OK (but dumped 4x), CHR OK (but dumped 2x), needed manual fixes to remove duplicates and adjust header

Cheesemeister3k, 2022-11-22:
The plugins provided do not provide coverage for all VRC mapper variants.
I've adapted the existing plugins to cover all retail Famicom VRCs.
See the readme for details.
[Konami VRC.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/10068676/Konami.VRC.zip)

tjanas, 2023-06-25:
Please re-test with the latest JB released today https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

Cheesemeister3k, 2023-06-25:
After updating to 6.7, I tested Gradius II with both the included "Konami VRC4.BIN" and "Konami VRC2c 4b.BIN" in the above "Konami VRC.zip" file.

"Konami VRC4.BIN" produced the issue described in the original post.
"Konami VRC2c 4b.BIN" produced a correct dump.

A binary file comparison of Konami BIN files in 6.6 and 6.7 shows no difference.
This issue is not fixed as of 6.7.

If you want to dump Konami VRC-based titles, please use the plugins from "Konami VRC.zip" above.

tjanas, 2023-06-25:
@Cheesemeister3k please copy this ticket to https://github.com/analoguejb/Analogue-Mega-Sg-JB so it can be addressed with the next JB update