SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #114: Feature Request: TurboGrafx-16 core

mikoyv3, 2021-04-10:
Because both the Analogue NT Mini Noir and upcoming Analogue Duo have Cyclone V FPGA. It would be amazing if the TurboGrafx-16 core be made available on the  NT Mini Noir

tjanas, 2021-04-10:
I would strongly bet against this happening for various reasons.

wiselinkpro, 2021-04-14:
this would be very great, honestly.  I just don't know if the author of the jailbreak has a PCE/TG16 core for this FPGA or not is the thing.

PizzaCube, 2021-04-29:
i would take it...  

but i am just going to buy the FPGA turbo duo they are making anyway...   So i dont think they are going to do that.  i am shocked there is a sega on the nt mini that i never use because i have a mega sg....

closed 2021-04-30 by tjanas