SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #116: SMS core in 6.5/6.6 results in black screen after loading Genesis/GG ROMs

jbaconbits, 2021-04-14:
I had this problem after updating to the latest jailbreak firmware/file structure folders, as did others in the classic gaming discord. 

Replacing the SMS core file to the older version from the original file structure download works fine.

prestia, 2021-04-22:
I came to report this same issue but didn't know about the workaround (thanks, @jbaconbits). Until there's a long-term fix, here's a step-by-step for the workaround:

1. Download “SD.Card.System.Structure_verJB6.2.zip” from here: [https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/releases/tag/v6.1](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/releases/tag/v6.1)
2. Mount your SD card with the 6.5/6.6 jailbreak
3. On the SD card, replace `SYSTEM/CORES/SMS.NTS2` with `SD/SYSTEM/CORES/SMS.NTS2` from the file you downloaded in step 1.

Genesis games should load fine now.

Ulisses-Medeiros, 2021-04-24:
To complement the games freeze and the option to access the jailbreak menu continues, however the games freeze after a while, and as reported above some only have a black screen, and some games like sonic show error screen. 

the games were obtained from the no-intro pack and corresponded to the CRCs correctly, without errors

Genesis, Game Gear, SG1000, SMS games, only work with the workaround of change the SMS.nt2 core from 6.5 or 6.6, by Jailbreak 6.2

Videos & Photos with Demonstrations:
https://drive.google.com/drive/folders/1JrWDw-6iqUtLobxVrwPLtnrIYGd1hlf7?usp=sharing

ZReport, 2021-04-26:
Here's two videos that show this issue:

https://www.youtube.com/watch?v=fEjJZBshJzY

https://www.youtube.com/watch?v=AygPj6cWjEI

BCProgramming, 2021-05-01:
I've also experienced this issue, and tried everything to address it (Fresh ROM sets, completely recreated SD Card, verified core files, etc). 

My experience was that pretty much any Genesis game would hang/freeze after a time. I found Sonic 1 would display an error (illegal instruction or address error typically) and sometimes I could force-resume with C (select). Other titles would trigger what seemed like anti-piracy protection- Streets of Rage 3 would arbitrarily transition to the title screen during gameplay. (And it too, would eventually freeze). Master System and Game Gear Titles I didn't see freeze but they would tend to get graphically glitched out, with corrupted palettes and nametables.

Reverting to 6.2 as described here has fixed the Sega system cores for me as well.

MadeBetter, 2021-05-02:
I had the same problem but after renaming my genesis rom extension from .smd to .md it works fine now, I didn't have to revert to previous 6.2 core

tjanas, 2021-06-01:
> I had the same problem but after renaming my genesis rom extension from .smd to .md it works fine now, I didn't have to revert to previous 6.2 core

@jbaconbits can you try the above? All verified no-intro Genesis roms should have .md file extensions.

closed 2023-06-26 by tjanas