SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #119: GBC Konami GB Collection 2&4 not working

AbnormalFreak, 2021-04-24:
This is for the PAL Game Boy Color versions of the series. Collections 2 and 4 don't work in the GBC core. They display the GBC startup screen but then get stuck on a blank white screen. Konami GB Collection Vol.4 (Europe) includes the color version of Castlevania II: Belmont's Revenge, which would really be cool to play. I've tried ROMs from two different sources and neither work. Every other GBC ROM I've thrown at it (including Konami GB Collection Vol.1 and 3) works great so I don't know why these don't.

tjanas, 2021-04-30:
Provide checksums of your files.

burbruee, 2021-07-02:
I can confirm this issue. This bug is present both on the old Nt Mini as well as on the Noir jailbreak.
It's odd that Vol 1 boots, they use the same mapper (MBC5) and I compared the header and they're identical.

The game works on the GB core as it is backwards compatible (black cartridge), but then of course you don't get the color as well as the advantages of running on GBC hardware. They optimized the games for GBC hardware and will have no lag in game. (at least Belmont's Revenge, can't speak for the other games)

md5: f3414d53473e2cc43347774cc5f40495  Konami GB Collection Vol.4 (E) [C][!].gbc

matthargett, 2021-12-25:
Confirmed to be reproducible on JB 6.6. Even the Konami GB Collection “speed hack” variants in the Hacks directory of the SmokeMonster pack are affected.

closed 2023-06-26 by tjanas