SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #120: Vice : Project Doom - jumpy status bar.

wezlyons, 2021-04-24:
During the side scrolling levels, the status bar at the bottom of the screen jumps up and down.

[image]

Ulisses-Medeiros, 2021-04-24:
Disable the option of 16 Sprites per Scanline, some games have problem with this option ON.

wezlyons, 2021-04-24:
Thanks for the reply.
I have tried this but unfortunately the issue still persists.

tjanas, 2021-04-24:
Is this a rom or cartridge? If the former, can you confirm that the NES rom with correct NES2.0 header has file checksum CRC32: ce8e8a9c

Mark-VI, 2021-04-24:
I have the same issue. I've tested both with a ROM and with an authentic Vice cartridge.

wezlyons, 2021-04-24:
I can confirm it is a rom with correct header and file checksum.

tjanas, 2021-04-24:
Might be a regression bug, I am still on the previous version, will try to test on that later

BaldWookiee, 2021-04-24:
Seems to be the same issue with Bram Stokers Dracula.

wezlyons, 2021-04-25:
The exact same issue occurs in Wolverine too.

AbnormalFreak, 2021-04-25:
This happens with ROMs for both Flintstones games also (Rescue of Dino & Hoppy and Surprise at Dinosaur Peak) even with 16-sprite option turned off.

tjanas, 2021-04-25:
I tested Vice on JB6.5 and there are no issues like this.

Ulisses-Medeiros, 2021-04-26:
> I tested Vice on JB6.5 and there are no issues like this. 

this is the version of the jailbreak i am using and the same problem mentioned above occurred with me if using the option of 16 sprites per scanline, If it is off it becomes normal

Mark-VI, 2021-04-26:
> 
> 
> > I tested Vice on JB6.5 and there are no issues like this.
> 
> this is the version of the jailbreak i am using and the same problem mentioned above occurred with me if using the option of 16 sprites per scanline, If it is off it becomes normal

On 6.6 it does this on the real cart or ROM, 16 sprites on or off.

tjanas, 2021-04-26:
> > I tested Vice on JB6.5 and there are no issues like this.
> 
> this is the version of the jailbreak i am using and the same problem mentioned above occurred with me if using the option of 16 sprites per scanline, If it is off it becomes normal

Are you sure you properly updated the firmware and the core files (from the latest SD Card System Structure zip) when you updated to 6.6?

Mark-VI, 2021-04-26:
Yep, I’m sure it properly updated. I’m on 6.6, not 6.5.

 

Are you sure you properly updated the firmware and the core files (from the latest SD Card System Structure zip) when you updated to 6.5?

tjanas, 2021-04-26:
This is confirmed to be a regression bug.

PizzaCube, 2021-04-28:
I got the same thing in a few games most have been mentioned except 

summer carnival 92 famicom

and some RPG game i was playing for like 10 minutes and was going to go back to later....  i cant remember the name..

PizzaCube, 2021-04-29:
the famicom version of the flintstone games do not have this issue for some reason.

tjanas, 2021-04-29:
The Famicom version use a different mapper:
```
	<game><!-- Licensed Japan\The Flintstones꞉ The Rescue of Dino &amp; Hoppy.nes -->
		<prgrom size="131072" crc32="547E6CC1" sha1="0609D3FA46BDE957636AFE28645422A52231F676" sum16="E1C8" />
		<chrrom size="262144" crc32="EB84A997" sha1="7D4C7E9A18F769873AB1D6AC9DC6AA455EDAD87A" sum16="209F" />
		<rom size="393216" crc32="40C0AD47" sha1="F949FC16932808C426074AB1628F246B0435C52C" />
		<pcb mapper="48" submapper="0" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```
```
	<game><!-- Licensed North America\The Flintstones꞉ The Rescue of Dino &amp; Hoppy.nes -->
		<prgrom size="131072" crc32="0AF937D1" sha1="0F7AB0E7F31F0C5863FEF6DDBE28080A5D3E7E8C" sum16="849B" />
		<chrrom size="262144" crc32="DC53CEFA" sha1="1E33C0A68140E74BCDE71F9AF5B9BBEF9B2181F6" sum16="D133" />
		<rom size="393216" crc32="2FE20D79" sha1="AD99C84C67659AF576A57CB500B8DEE82ADED7DD" />
		<pcb mapper="4" submapper="0" mirroring="H" battery="0" />
		<console type="0" region="0" />
		<expansion type="1" />
	</game>
```
On a slightly related note, the Famicom required a hack to the correct NES header for it to work on the original Nt Mini (had to change it from 48 to 33), as documented [here](https://github.com/tjanas/Nt-Mini-Legacy-Jailbreak/issues/2), since Kevtris has not fixed any of the outstanding bugs on the original Nt Mini since 2017.

tjanas, 2021-04-30:
> I think this is related, let me know if I should open a separate bug report.

Please make separate bug reports for these two games so they don't get forgotten.

tjanas, 2021-04-30:
> I think this is related, let me know if I should open a separate bug report.
> 
> Crystalis (1bd39032) the whole screen jumps when there's dialog box. The dialog box and status bar remains stable. I included a gif of this [here](https://www.reddit.com/r/AnalogueInc/comments/n0y2bz/crystalis_screen_bouncing_during_dialog_box_on/)
> 
> Summer Carnival '92: Recca (d98ec487) the lower status bar jumps
> 
> Both are resolved when I set PAL region but both games are NTSC. I'm running JB6.6 firmware

Your Crystalis rom is bad; it should have the following file checksums:

Size: 393232 bytes
CRC32: F8F1865D
SHA256: 9BEC6E35D1174F43A32C2DB2BE972A1627D04F3613EBB4246ADD2C2B4BE0747F
SHA1: C9144B473AC2A3B9BF3EA8054AFF565460E5B2D4
MD5: 7F305A2C8E55806C29BF7CEB036E83BD

I'm guessing you didn't use a rom that matches the [Nt Mini Noir HTGDB](https://github.com/frederic-mahe/Hardware-Target-Game-Database), or you didn't use the NES 2.0 XML Database and the NES Header Repair Tool Python Script to fix your headers (see the [readme](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak)).

tjanas, 2021-04-30:
@PizzaCube please confirm your Vice : Project Doom rom has this file checksum:
```
Size: 262160 bytes
CRC32: CE8E8A9C
SHA256: 4CE488710ACC6BA9ABA812C69AB7A7C2A08587F9FEF95EC6E5A67B50792E2CF2
SHA1: 36B9DFA6FA60652CA0AD15E7A5C13192B62606BC
MD5: 34C954E49BA8E7F410FC3D76AD01ECE0
```

tjanas, 2021-04-30:
> Summer Carnival '92: Recca (d98ec487) the lower status bar jumps

Your file is bad. Correct file has the following checksum:
```
Size: 262160 bytes (256 KiB)
CRC32: 48A42E2F
SHA256: B4197F6D0BB517D5144BFE799E1C03C9FF07048B72884A4487821F1711A59D25
SHA1: 7E8506E047699BDEF04A77C2CF50335BE8133121
MD5: D1977209742B71A93E4EBC1AFA0063DC
```
Per HTGDB:
```
b4197f6d0bb517d5144bfe799e1c03c9ff07048b72884a4487821f1711a59d25	Analogue Nt Mini Noir/NES/2 Japan/Japan P-S/Summer Carnival '92 - Recca (Japan).nes	7e8506e047699bdef04a77c2cf50335be8133121	d1977209742b71a93e4ebc1afa0063dc	48a42e2f
```

nocmonkey, 2021-04-30:
Thanks @tjanas!  I overlooked those instructions on the readme.  I patched both the roms using nes_header_repair.py and nes20db.xml:

crc32:
f8f1865d	Crystalis (USA).nes
48a42e2f	Summer Carnival '92 - Recca (Japan).nes

md5sum:
7f305a2c8e55806c29bf7ceb036e83bd  Crystalis (USA).nes
d1977209742b71a93e4ebc1afa0063dc  Summer Carnival '92 - Recca (Japan).nes

I can confirm that the jumping issues are still present on both roms.

tjanas, 2021-04-30:
[Video](https://www.youtube.com/watch?v=ZAxoak6Skf8) of Vice: Project Doom with a real cart

PizzaCube, 2021-04-30:
> @PizzaCube please confirm your Vice : Project Doom rom has this file checksum:
> 
> ```
> Size: 262160 bytes
> CRC32: CE8E8A9C
> SHA256: 4CE488710ACC6BA9ABA812C69AB7A7C2A08587F9FEF95EC6E5A67B50792E2CF2
> SHA1: 36B9DFA6FA60652CA0AD15E7A5C13192B62606BC
> MD5: 34C954E49BA8E7F410FC3D76AD01ECE0
> ```

it is the same and my real carts of the games in this thread do the same thing.

Mark-VI, 2021-05-01:
> 
> 
> I got the same thing in a few games most have been mentioned except
> 
> summer carnival 92 famicom the sound is also wrong
> 
> and some RPG game i was playing for like 10 minutes and was going to go back to later.... i cant remember the name..
> 
> days of thunder
> 
> mega man 5

I can confirm that Mega Man 5 does this on all of the stage intro screens. I'm using an original cart, not a ROM. I can provide a video for confirmation if it helps.

PizzaCube, 2021-05-01:
Klax tengen version is mega broken the worst of them all. the entire screen is moving

PizzaCube, 2021-05-03:
the other game that has this issue is  taxan magician

nocmonkey, 2021-05-16:
> I tested Vice on JB6.5 and there are no issues like this.

I flashed JB6.5 and can also confirm that there are no issues with Crystalis and Recca on this build.  The jumping problems look to have been introduced in JB6.6

BobsYurUncle, 2021-09-05:
Can confirm a regression, Kickmaster authentic cartridge has a jittery status bar on JB6.6 but not on JB6.5, regardless of dejitter/16-sprite settings.

PizzaCube, 2022-01-30:
i just got an ever drive n8 pro  and it does this with all the jumpy games with real carts and the n8 pro

nobleEightfold, 2023-03-08:
As far as I can tell, all of the US games mentioned here are MMC3, so there could be a regression the PPU A12 address read/write timing, which is snooped by MMC3 to drive its scanline/IRQ counter, which is used commonly for things like drawing status bars at specific parts of the screen.

BaldWookiee, 2023-03-08:
Would be nice if Analogue would look into this. It's only been 2 years. =/

AbnormalFreak, 2023-03-09:
> Would be nice if Analogue would look into this. It's only been 2 years. =/

I did just get this email from Analogue in response to wondering when the firmware will be updated to address this issue:

At this time, we do not have a specific date or timeframe for when the new firmware will be released for Nt Mini Noir. This issue will be addressed in a future update.

Thank you very much for your continued patience with us.

Best Regards,
Analogue Support

tjanas, 2023-05-04:
@wezlyons @Mark-VI @PizzaCube @BaldWookiee @AbnormalFreak @BobsYurUncle please re-test your cartridges with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

BobsYurUncle, 2023-05-05:
I retested Kickmaster and Vice Project Doom on the firmware released today.

Kickmaster's status bar seem normal however Vice Project Dooms status bar is still jumpy when 16 sprites per-scanline is enabled.

closed 2023-05-05 by tjanas

Mark-VI, 2023-05-05:
I can confirm that MM5 is fine as is Vice (as long as you don't enable 16 sprites).