SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #121: Core Loading Error! on all cores except NES

1kimota, 2021-04-26:
I have a feeling this is due to a Bios issue but the README.txt tells me to look in Bios folder for instructions and the Bios folder tells me to look at the README.txt, is there something I am missing?

eltees, 2021-04-26:
Bios files should be present in the /BIOS/ folder with the default names mentioned in the release notes for each core, though typically you can also manually go to the settings after loading a given core and open the file browser to select a specific bios there to over-ride the default (useful if you want to swap between the 'normal' GB bios and the faster loading / non scrolling super gameboy bios for the gb core, etc)

1kimota, 2021-04-26:
Sorry to sound frustrated but the /bios/ folder only has one file, a readme that states the bios info has been moved to the README.txt, the README.txt states to look in the bios folder.  The README.MD states instructions listed below with nothing below.

eltees, 2021-04-26:
You need to track down the bios files yourself as they are not included in the jailbreak.

Once you have sourced the bios for a system you would like to try, you should be able to find the default file name for it (often slightly different) rename it, and then place it in the /BIOS/ folder of the sd card.

1kimota, 2021-04-26:
Alright so I did that and sorry for my ignorance but how do I rename it.  I am unsure what you mean by "default file name"?

eltees, 2021-04-26:
Each core lists in its release notes the default file name it is intending to load from the /BIOS/ directory, so for example from the notes on the Atari 7800:

This core needs a BIOS to run which you can select with Core Options. The Core will load the file 7800bios.bin (expected CRC32: 5D13730C) found in the /BIOS/ directory by default.

So if your atari 7800 bios is named something different, you can either rename the file to match the expected default name, or you can go to the settings after loading that core and manually select it to load the other filename if you prefer.

1kimota, 2021-04-26:
ah so maybe not a BIOS problem core will not load at all, if I choose NES core it brings me to file navigation menus etc but if I choose Genesis it gives a core loading error!

eltees, 2021-04-26:
In following the install guide on the main page, did you extract the sd card file structure for step 2 from the zip file to the root of the SD card? That should have placed all of the necessary data in the right places to launch each of the provided cores.

1kimota, 2021-04-26:
Yes, I am able to run NES games fine.  All the others give a core loading error!

1kimota, 2021-04-26:
Also thank you for looking at this with me! Your time is appreciated!

1kimota, 2021-04-27:
The  folder that was unzipped to root needed to be deleted. The files needed to be in root. Problem solved!

closed 2021-04-27 by 1kimota

MadeBetter, 2021-05-02:
I updated the instructions for this and created a pull request, because many people seem to have the same issue due to misleading instruction.