SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #123: Console locks up? and does not show video when changing controller ports.

PizzaCube, 2021-04-29:
So i have been plugging in 1 controller into each of the controller ports. taking it out of player 1 and putting it in the 3 other slots while a nes game is running. (cart or rom any game) any controller type real or  the included wireless.

the console locks up when you do this on the new firmware.  the screen will go  black and the hdmi monitor goes into standby mode.  so i cant tell if it crashed or just video dropped.

i tried this with the 4 player option check on and off it does not seem to matter

tjanas, 2021-04-29:
I don't think you should be plugging/unplugging non-USB controllers on any video game console while it is powered on.

PizzaCube, 2021-04-29:
the real nes you can do this with out issues

tjanas, 2023-05-04:
@PizzaCube please re-test with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

closed 2023-06-26 by tjanas