SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #124: The Legend of Zelda Vertical Dotted Lines

codyjarvis, 2021-04-30:
Vertical dotted lines appear on screen when playing two separate copies of The Legend of Zelda.
Firmware: Nt-Mini-Noir-Jailbreak v6.6
Game: The Legend of Zelda - NES-ZL-0 PRG
Issue: Vertical dotted lines.

Tests/attempted fixes:
Cleaned both carts and tested continuity. Visually everything looks great.
Tested both carts on original hardware with no issues.
Dumped roms using copy nes. Roms looked great - no vertical dotted lines.
Flashed original firmware to Nt-Mini. Vertical lines no longer appear.

codyjarvis, 2021-04-30:
Also attempted with dejitter both on and off in case the issue was related to issue #95

PizzaCube, 2021-05-07:
my gold and grey cart does not have this issue so i am not sure why yours does.

codyjarvis, 2021-06-09:
@brym103 do the vertical lines disappear on the stock firmware? Mine do

brym103, 2021-06-09:
> @brym103 do the vertical lines disappear on the stock firmware? Mine do

Just tested -- no, I get the same issue when I flash back to the stock version 1.2 firmware.

brym103, 2021-06-09:
Update - I do NOT get the issue when I reverted to the stock version 1.1 firmware.  The cart plays perfectly on version 1.1.  I'm going to post this as a separate issue, but note that it may be a similar issue as this one.

tjanas, 2021-06-09:
Have you contacted Analogue directly and via Twitter? Something like this should be brought to their attention.

brym103, 2021-06-09:
> Have you contacted Analogue directly and via Twitter? Something like this should be brought to their attention.

I have not.  I was under the impression that posting issues here was the best way for the right people at Analogue to see them.  Is that not the case?

tjanas, 2021-06-09:
Perhaps the jailbreaker looks here from time to time, but this is not official by any means. Problems with official firmware and carts should definitely be brought to the attention of Analogue on Twitter, especially since it would pertain to their public relations and claims of perfection. Perhaps other users on Twitter with these issues could also chime in.

tjanas, 2021-06-09:
I think bringing it up on Twitter would help light a fire under Analogue's support teams and help direct their priorities to fixing issues like these (they may be unaware of all of these issues).

brym103, 2021-06-09:
Fair enough; I'll send a tweet their way.

whitezombie2000, 2022-02-13:
I cannot reproduce this issue with a real cartridge. I tested a gold Rev A copy of Zelda on the 6.6 JB and 1.2 firmware. Plays fine for me and so does the dumped ROM.

tjanas, 2022-02-14:
I’m confused why many people have experienced this issue while others have not with JB6.6. I don’t recall these issues reported with JB6.5. Are we sure it should be closed?

whitezombie2000, 2022-02-14:
Here's my cart of Zelda. I can't believe that battery still works. I need to replace that soon!

[image]

tjanas, 2022-02-14:
My battery (and save file) on my Zelda 1 cart from 30+ years ago are still working too. :) Those Maxell batteries were great.

cizzymac, 2022-12-08:
Leaving a comment here to advise that I receive the same issue with vertical dotted lines with my Everdrive N8 Rev 1.4, as well as my official Metroid Rev A cart, on JB 6.6. Flashed my Noir to official firmware 1.2 and get the same issue. Flashed to 1.1 and the lines are gone. Reflashed Noir with JB 6.6 without disturbing my seated Everdrive cart and the vertical dotted lines have returned. This is definitely a firmware issue causing this problem.

tjanas, 2022-12-08:
@cizzymac Please contact Analogue support to encourage them to release an update, and it wouldn’t hurt to let them know on Twitter too.

tjanas, 2022-12-08:
Have you tried flashing to JB 6.5? That seems to be the best fw at this point.

cizzymac, 2022-12-08:
@tjanas I did reach out to Analogue both via Twitter and via their support email on their website. It will be a shame to lose out on the fixes provided on 6.6, but I'll give 6.5 a shot today and see how it goes. Thanks much for your prompt response!

tjanas, 2023-05-04:
@cizzymac @codyjarvis please re-test your cartridges with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

cizzymac, 2023-05-04:
@tjanas I can confirm the issue is resolved with the new official firmware on my end.

closed 2023-05-04 by tjanas

whitezombie2000, 2023-05-04:
@tjanas Should this be closed yet? The OP has not confirmed if their copy of Zelda is working. @cizzymac Never reported an issue with Zelda. Or can @cizzymac confirm that Zelda was an issue before and is working now.

tjanas, 2023-05-05:
@codyjarvis could you test please?

cizzymac, 2023-05-05:
@whitezombie2000 My issue was with Everdrive N8 1.3 and Metroid. I do not have a LoZ cart to test for you unfortunately.

codyjarvis, 2023-05-05:
I can load the new firmware and test this weekend.

whitezombie2000, 2023-05-13:
@codyjarvis How did the test go?