SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #126: Rockman 5 jittery elements

MrFilz, 2021-05-01:
In the latest FW 6.6 Rockman 5 has not correct working graphical elements. The lower half of the boss introduction screen, or the elevator at the beginning of Gyro Man stage are kind of jittery. It happens with my original cart and with the rom played via jailbreak.
I would like to ask a question, it may be stupid. Is it intended behavior, that there are two lines of pixels completely cropped on every side of the screen in pal mode? I have a huge pal collection.

PizzaCube, 2021-05-01:
yes rockman 5 boss screens have it.

so does mega man 2 maybe in the mini boss when you transfer over to the next screen it will jump once and i dont remember a real USA or japan nes doing that.

adam16bit, 2021-05-14:
I have also noticed jittery elements with Summer Carnival Recca and Bart vs. The Space Mutants in the score/menu areas on the bottom of the screen.

MrFilz, 2021-06-13:
I tested a bit more. The broken graphical elements of Mega Man 5/Rockman 5 are complete normal when playing in pal mode. I played my pal copy recently and switched to pal mod with 1080p50 for this, and voila, everything seems to be normal. This works with the us or jp version of the game too. Of course nobody wants to play these versions in pal mode... Maybe this helps.

acidrain97, 2021-10-30:
Got a physical copy of Mega Man here are the results.

On current jailbreak firmware.

[image]
[image]
[image]

tjanas, 2021-10-30:
This is also an issue with the latest official firmware too, correct? Has anyone contacted Analogue support regarding this bug?

MrFilz, 2021-10-30:
They do not care about the issues with the latest firmware. It is 8 month
now since the last update... and the most problems are not present in the
previous firmware. So it can't be that hard to give us a working one right?
This was a very expensive console and also my last one from Analogue! I do
not like how they handle the support of their products!

tjanas, 2021-10-30:
no issues with Mega Man 1 cart with latest firmware here:

https://twitter.com/ecuprofessor/status/1454503214045356035?s=21

acidrain97, 2021-10-30:
Can you give me an idea of what setting you're using? PAL, NTSC? This is the first cart I've found to be wonky. The disk System stuff is so screwed up. To be honest the jailbreak roms work better than the physical stuff. 

But yeah I'm not happy Analogue on this.

tjanas, 2023-05-04:
@MrFilz @acidrain97 please re-test your cartridges with the latest official firmware update released today https://www.analogue.co/support/nt-mini-noir/firmware/1.3

acidrain97, 2023-05-04:
If I update I'm guessing it will no longer be jail broken and all my setting for various cores go back to default, correct?

tjanas, 2023-05-04:
That is correct; you would have to make notes (take screenshots) of your settings when you change firmware versions, including to & from official and JB.

acidrain97, 2023-05-04:
When I see a jailbreak I'll test both... curious to see how much has been really fixed. I have find my backup settings first before I do anything.

tjanas, 2023-05-04:
Just so you are aware, you will also have to back up your settings when you update to a different JB version. I want to make sure this issue is fixed in both the latest official firmware as well as a future JB firmware.

acidrain97, 2023-05-05:
When I get home I'll give it a go... I have a pocket so no big deal. All the cores are on that anyway... I need to test other things as well. 

THX...

acidrain97, 2023-05-05:
[image]

acidrain97, 2023-05-05:
Sorry for the mess (one of my many test benches) YES Mega man 1 is fixed. No more lines...
[image]

Sorry for the mess... Mega Man 1 is fixed. No more lines. Tested FDS, both drive/fdsstick ran perfectly fine with the dejitter off. I read somewhere Legend of Zelda had issues, ran fine. Job Well Done!

tjanas, 2023-05-05:
Just need confirmation from @MrFilz or someone else with a Rockman 5 cartridge.

acidrain97, 2023-05-05:
What about a mega man 5 cart? I'm guessing they are the same game? I  might be able to get my hands one just not the famicom version. No guarantees.

MrFilz, 2023-05-05:
I will test the PAL cartridge and the famicom cart of Mega Man 5 as soon as 8 am Back at home from work😉

acidrain97, 2023-05-06:
[image]

I believe Rygar was on the release notes. Real cart runs fine. If moving forward the background tends to dim, and when static the background regains it's color. Does this on a lot of games that have vertical scrolling. SMB is one that comes to mind. Minor annoyance. Probably because of the screen I'm using. 

BTW I don't have Castlevania but VS. Castlevania from Retrozone/Retrousb work fine. Any game with dip switches work fine.

Squashing bugs like crazy... This is very good.

MrFilz, 2023-05-06:
I tested the PAL and NTSC-J cartridges of Mega Man 5/Rockman 5 and can confirm, all the jitter is gone! The everdrive n8 is working too without problems! They fixed a lot of bugs without introducing new ones so far. More than two years after the last firmware release i am happy now with my 500 Dollar purchase. Better late than never i guess...

acidrain97, 2023-05-06:
Jailbreak it and I'll be real happy.. just don't break anything in the progress.

closed 2023-05-06 by tjanas