SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #127: Core loading error !

MadeBetter, 2021-05-01:
Jailbroken Nt mini noir v2 won't load any core except for the NES one. 
When clicked on any other core the message below will be displayed
[image]

- All necessary bios files have been added with the correct names. 
- Tried using different SD card
- Made sure SD card is formatted using Fat32
- No SD cards exceed 32Gb
- Tried flashing back to original firmware then back to Jailbreak 6.6 (five times already)
- followed all instructions per readme.txt

nothing works!! 😔

tjanas, 2021-05-01:
Are you using the SD Card System Structure for v6.6? https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/blob/main/firmware/SD%20Card%20System%20Structure_verJB6.6.zip

codyjarvis, 2021-05-02:
@MadeBetter i had this same issue. The instructions are a bit misleading when they tell you to not change the file structure. Try moving all of the folders out of the "SD Card System Structure" folder and directly on the root of your sd card.

MadeBetter, 2021-05-02:
> Are you using the [SD Card System Structure for v6.6](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/blob/main/firmware/SD%20Card%20System%20Structure_verJB6.6.zip)?

Yes of course!

MadeBetter, 2021-05-02:
> @MadeBetter i had this same issue. The instructions are a bit misleading when they tell you to not change the file structure. Try moving all of the folders out of the "SD Card System Structure" folder and directly on the root of your sd card.

The instructions are misleading indeed, after putting the actual content of the "SD Card System Structure" in the root now it is working fine. The instructions needs to be updated! Thank you very much @codyjarvis

closed 2021-05-02 by MadeBetter