SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #130: SN30 Bluetooth using NES retro receiver incorrectly mapped??

1kimota, 2021-05-02:
As stated in title I am using a Bluetooth SN30 with retro receiver but is still mapped C button to select. Anyone have a fix or work around to at least not have C button mapped to select?

PizzaCube, 2021-05-02:
the receiver seems to have a usb port for firmware updates try that. i do not own that one but i have retrobit sega controllers that you can upgrade the firmware and they add new button mappings.

1kimota, 2021-05-02:
I tried a few different firmware updates and the Analogue firmware for the mini(not NT Noir). Thanks for help maybe they will update firmware for Noir too.

1kimota, 2021-05-08:
Yeah I tried the firmware and it states not for use with Noir, when using that firmware the receiver does not connect to SN30 bluetooth no matter what 8bitdo firmware I roll back to(I tried all available) Thank you for your assistance but I think I have to wait for an Analogue Mini NT Noir custom firmware update.

eltees, 2021-05-08:
I'm using the custom firmware with a retro receiver on my noir, but on an original F30 pro pad.  I think the issue you may be running in to is not a problem with the noir, but that your pad may be newer than the custom firmware can recognize (anything made after ~2018ish may have issues).

You could try the different bluetooth modes on the pad and try re-pairing it with each.  I have found it to work best with the d-input mode, but YMMV.

1kimota, 2021-05-08:
I will def try that! And thanks again for your help!

closed 2021-05-08 by tjanas