SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #131: Game Boy and GBC bios not loading.

Gearkeeper, 2021-05-03:
i have the bios files in the right file location but the bios will not load for either GB or GBC please help

PizzaCube, 2021-05-03:
are they named correctly?

Gearkeeper, 2021-05-04:
Yes just like the readme

tjanas, 2021-05-04:
> Yes just like the readme, but is it possible that i have the wrong bios for the NT noir ??

Does the checksum of the [GBC BIOS](https://datomatic.no-intro.org/index.php?page=show_record&s=47&n=1375) match what's listed in the readme?
```
You must select the Gameboy Color's bootstrap (BIOS) for it to work. gbcbios.bin (expected CRC32: 41884E46) is the default file to be loaded if found in the /BIOS/ directory.
```

Gearkeeper, 2021-05-04:
ok so this is where i dont understand what you me by bootstrap. im new to this stuff, is there a video that shows step by step how to do this because im lost

Gearkeeper, 2021-05-04:
also when i click the link "GBC BIOS" i get this,,,,,,,, A silly bug in code may have banned you with reason: Wrong URL.
Error id: 325.
I'm not bad. I'm just drawn that way.
Contact me for support: shippa6@hotmail.com

tjanas, 2021-05-04:
> im new to this stuff, is there a video that shows step by step how to do this because im lost
Folks in this [discord](https://discord.gg/UDu5ztY) may be able to help you more effectively

closed 2021-05-04 by tjanas