SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #133: Famicom Disk System won't load/play games on newest jailbreak (6.6)

isfet0, 2021-05-11:
I recently came into possession of an NT Mini Noir and then followed that up with a purchase of the Famicom Disk System. When the FDS arrived, I was unable to get it to load games properly; it would load the FDS launch screen (Mario and Luigi running around) and then would only sometimes load the title screen of the game before crashing entirely.

At first I wasn't sure if it was the Noir, the FDS itself, or the disk, but after trying a few disks and playing with the settings on the Noir, I ultimately gave up.

UNTIL a friend suggested I try to downgrade my jailbreak/firmware to version 6.2, and **suddenly** the FDS was completely functional!

So yeah, the FDS/cartridges in general (according to that friend) seem to potentially have some issue on the latest version of the firmware/jailbreak.

Sorry if this has been reported/recorded elsewhere, but I spent about 2 weeks combing the internet looking for info on this and found nothing, so I figured I'd post about it.

tjanas, 2021-05-11:
Can you ask in the #analogue-help Discord channel [here](https://discord.gg/UDu5ztY) and report back with the cause/solution?

tjanas, 2021-05-11:
If you are using a real FDS, does it work with the latest official firmware? If not, then it’s not a specific issue with the JB. Maybe you don’t have fully buffered mode enabled which is required for FDS when using HDMI?

isfet0, 2021-05-11:
OK, so!

After checking in with the recommended Discord channel, they said that I needed to Enable Dejitter in order to use the FDS with v6.6 and, what do you know, it worked.

So it seems like the options are to either use an older firmware, or to Enable Dejitter, and that will take care of the issue. Would obviously be ideal if Analogue fixed the issue entirely, but this is a good alternative, for now :)

Thanks for the recommendation, @tjanas !

tjanas, 2023-05-06:
@isfet0 can you please re-test with the latest firmware?

tjanas, 2023-06-26:
@isfet0 Please re-test with JB6.7 and if it is still an issue, open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB. Per the JB6.7 changelog:
- Fixed Famicom Disk System dejitter setting compatibility issue (Not compatible with Zero Delay)

closed 2023-06-26 by tjanas