SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #136: Cartridges Work on Official Firmware 1.2, but Not on Jailbreak 6.6

adam16bit, 2021-05-14:
I have 3 recently released cartridges that work perfectly on the official firmware, but glitch out on the Jailbreak preventing advancement of more than a few second into each game.

The ROMs of these games work fine on the jailbreak - but not the actual cartridges.  I updated back and forth to verify this. The cartridges are all clean (and new), so the only variable changing seems to be the Jailbreak Firmware causing them to not run.

I've tried cleaning (and re-cleaning, and re-recleaning) the cartridges to no avail.

I've tried the other voltage settings in the menus, and nothing changes.  The only variable change that gets these cartridges to run without glitches 100% of the time is by going back to the official firmware.  The games DO work on this hardware, just not with the Jailbreak.

**From Below**
Title screens work fine.  After I press start, I get random game overs. https://twitter.com/Adam16bit/status/1393005974542184449
This is only with the Cartridge on the Jailbreak.  The ROM on the Jailbreak works fine.  The Cartridge on Original Firmware works fine. The cartridge works perfectly with my AVS or NES consoles.

**Micro Mages**
Titles work fine. After I press start I can play briefly, but random graphical glitches appear before the game freezes, goes blank, or I get a screen of random sprite junk.  This is only with the Cartridge on the Jailbreak.  The ROM on the Jailbreak works fine.  The Cartridge on Original Firmware works fine. The cartridge works perfectly with my AVS or NES consoles.

**Project Blue**
Titles USUALLY work fine.  The game glitches out after the game goes from the introductory sequence to a playable space. Sometimes it goes dark, sometimes the colors change and it glitches/freezes.  
This is only with the Cartridge on the Jailbreak.  The ROM on the Jailbreak works fine.  The Cartridge on Original Firmware works fine.  The cartridge works perfectly with my AVS or NES consoles.

I tried my other recent aftermarket games and so far have had no problems with the first level or two of:
Eskimo Bob, Jay & Silent Bob's Mall Brawl, Nebs N Debs, Haradius Zero, Doodle World

eltees, 2021-05-14:
Thats odd, my micro-mages cart works on the mini noir with both current jailbreak and official firmware (apart from the four score issues with the current NES core I documented in another bug report).  One thing potentially different in my case though based on what I saw in that post is I'm using HDMI outputs and 'Zero Delay' for the buffering mode, which i know runs things slightly differently than if direct analog outs are in play.

eltees, 2021-05-27:
Finally got the cables to hook it up over analog connectors just to see if the slightly changed timing impacts anything, and still no issue running micro mages via cart on current jailbreak noir on my end, played through first 3 worlds without problems so about 25-30 minutes in.

Have you tried using a more capable power supply? I've seen some rumors that the stock one sometimes has issues running certain carts as its right on the edge of the power delivery it can offer.  People on the discord seem to really like the triad magnetics WSU120-1500-13.  I've also been running off one from an old seagate external hard-drive that can deliver up to 3 amps

LoneBoco, 2021-06-09:
My Micro Mages also encountered similar issues.  Lots of graphical corruption.  Sometimes the graphics stop rendering entirely.

tjanas, 2023-05-04:
@adam16bit please re-test your cartridges with the latest official firmware that was released today.

closed 2023-06-25 by tjanas