SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #138: Graphical glitches with Kirby’s Adventure PAL and Everdrive N8 Pro

lesserkuma, 2021-05-23:
Kirby’s Adventure PAL on an Everdrive N8 Pro (latest OS) seems to be broken on the latest firmware. My console is set to PAL mode. I’ve tried the 3.6V Cartridge IO option and all buffer modes (fully buffered, zero delay, single buffer), but none of these fix it. No difference between old iNES and iNES 2.0 headers and it happens with all PAL localised versions (English, German, French).

[image]

Jeffersonian, 2021-12-20:
I can reproduce this issue with my N8 Pro on my Nt Mini Noir using firmware version 1.2 (non-jailbreak).

tjanas, 2023-05-04:
@Jeffersonian @lesserkuma can either of you re-test with today’s firmware update?

tjanas, 2023-05-06:
[Fixed](https://youtu.be/pPwWZ2D2Pjw).

closed 2023-05-06 by tjanas