SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #139: Feature Request: Mega Drive / Genesis button mapping

Elrinth, 2021-06-01:
I would very much like to play the Mega Drive core with the SNES controller via the NES->SNES adapter which is sold by Raphnet.
https://www.raphnet-tech.com/products/snes_to_nes_cable/index.php

Currently the button layout is totally weird and playing an extremly button-required game like XenoCrisis on the NT Mini Noir is difficulty increased by ten-fold as the button mapping makes no sense.

Please make it so you can in the menues select which button corresponds to what on the Mega Drive controller.

tjanas, 2021-06-01:
Changing button mappings has never been offered by any Analogue JB firmwares, including the one made by Kevtris for the original Nt Mini. Unfortunately I don’t think this feature request will happen.

eltees, 2021-06-12:
It can 'sort of' be done though over complicatedly by using the snes adapter combined with a snes retro receiver with current firmware combined with an 8bitdo pro 2 controller which allows you to manually remap the controller buttons to anything you would like so you can get a nice a-b-c layout across the front buttons, but yes I would very much appreciate if it would be possible to allow either direct remapping in the controller configuration or at least a button map text file (possibly per core) to at least have *a* way to do it without needing to buy three separate external devices for close to $100 to get there.

thomaskaelin, 2021-07-09:
I would like to point out the findings mentioned in #22. If there would be any way to configure the way a core reads the inputs received from the controller ports these problem / feature request could be addressed. Maybe reading the mapping from a configuration-file on the SD card?

eltees, 2021-07-11:
The problem mentioned in  #22 appears to be what I mentioned above about the retro receiver firmware.  The legacy analogue custom firmware for that device treats it as its own snes-input adapter which opens up more buttons.

Unfortunately the mapping still isn't ideal/perfect (though it is much better, and is the same mapping you will get with a wired SNES to nes adapter and an snes retro receiver).  But one big problem is that custom firmware for the retro receiver is based on an early 2017 firmware from 8bitdo, and those use bluetooth device whitelisting so will only talk to 2017 or earlier controllers.

That all said right now the 'best' universal input option for the genesis core with  consistent A,B,C face buttons natively and decent 6 button support is:

- SNES wired adapter plugged in to nt mini noir's first port.
- SNES retro receiver on current firmware plugged in to snes wired adapter
- 8bitdo pro 2 paired with snes retro receiver
- custom 8bitdo Pro 2 controller remap to give you A-B-C console input across the Y,B,A buttons and X-Y-Z button input across the L-X-R buttons

The M30 works in this configuration as well but its button mapping is a little messy matching the intended input noted in #22 (aka A-B-C console input on B,Y,A buttons respectively... not unplayable but certainly un-natural feeling.

tjanas, 2023-06-26:
Open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB if this feature request is still wanted.

closed 2023-06-26 by tjanas