SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #142: Rygar Cart - Vertical dotted lines on jb 6.6 firmware and stock version 1.2 firmware, but works fine on stock 1.1 fw

brym103, 2021-06-09:
When playing my Rygar cart, I am getting dotted vertical lines on the screen similar to what are described on Zelda in issue #124.  I get this issue when using both the jb 6.6 firmware and the stock 1.2 firmware.  The cartridge has been thoroughly cleaned.  

When using copyNES mini on the jb 6.6 firmware, I was able to dump a ROM that works perfectly on the jb 6.6 firmware.  The MD5 checksum for the rom matches the known good dump of Rygar (USA) (Rev 1).nes (after stripping the header).  

I reverted to the stock version 1.1 firmware, and the cart works perfectly there.

cmcginty, 2021-06-26:
Also confirmed with Castlevania original cart. After downgrading to jb 6.5 the vertical lines are no longer visible. I did not attempt to dump the ROM to verify, but will be happy to if necessary.

whitezombie2000, 2022-02-13:
I cannot reproduce this issue with a real cartridge of Castlevania and Rygar on the official 1.2 firmware or the 6.6 JB firmware. It should be noted that my Castlevania is Rev A as stated on the back of the cartridge and Rygar is not Revision A.  I can dump both carts with CopyNES using mapper UxROM.BIN, load both dumped ROMS with out issue on the 6.6 JB firmware and on an emulator.

tjanas, 2022-02-14:
@brym103 which Rygar cart do you have? Is it the original (https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=1874) with “ NES-RY-0 PRG” on the ROM chip inside the cart, or the revision cart with “NES-RY-1 PRG” on the ROM chip?

tjanas, 2022-02-14:
@cmcginty if you open your Castlevania cart, which ROM chip is it? “NES-CV-0 PRG” or “NES-CV-1 PRG” ?

tjanas, 2022-02-14:
[image]
[image]

brym103, 2022-02-14:
@tjanas I don't have the tool needed to open the cart at the moment, but when I dumped the rom, it matched the checksum for this version after I removed the headers:
https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=1875

For what it's worth, I eventually reported this issue directly to analogue support (maybe in September?) and they told me that they were aware of the issue and it would be fixed in the next firmware update.  Of course, who knows when that will be.

tjanas, 2022-02-14:
This issue should remain open until at least the release of JB6.7 or newer.

whitezombie2000, 2022-02-14:
Here's mine.

[image]

tjanas, 2022-02-14:
@whitezombie2000 I highly encourage you to submit that dump and PCB photo to no-intro, especially since they don’t have any photos for the initial version of Rygar. Or I can reach out to a No-Intro member to have your verification and photo(s) added on your behalf.

https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=1874

https://forum.no-intro.org/

whitezombie2000, 2022-02-14:
Here's my Castlevania 1 cart.

[image]

tjanas, 2022-02-14:
I see that is a very late (1989) CV1 cart. You should submit scans of that one too

whitezombie2000, 2022-02-14:
@tjanas You can reach out to them if you want too. I will provide the two dumps here and delete them after you grab them.

[Files.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/8057396/Files.zip)

cmcginty, 2022-02-14:
> @cmcginty if you open your Castlevania cart, which ROM chip is it? “NES-CV-0 PRG” or “NES-CV-1 PRG” ?

NES-CV-1 PRG
The chrram is a SHARP LH5164D3-L ... all on a NES-UN-ROM-07 PCB.

tjanas, 2022-02-14:
@whitezombie2000 your dumps and PCB photos have been passed along to a no-intro dumper and submitted.

tjanas, 2022-02-21:
@whitezombie2000 I have been informed they are now part of the No-Intro dat, credited to you:

https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=1874

https://datomatic.no-intro.org/index.php?page=show_record&s=45&n=0358

ManeFunction, 2022-10-20:
I assume, this is the same issue: https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/issues/148

brym103, 2022-10-21:
> I assume, this is the same issue: #148

Looks like it to me, yes.

cizzymac, 2022-12-08:
Just leaving a comment here as well that I am receiving the same exact issue with my Everdrive N8 Rev 1.4 and my Metroid Rev A cart. Vertical, dotted lines persist on JB 6.6, official 1.2 firmware, and are resolved with official 1.1 firmware.

tjanas, 2023-05-04:
@cmcginty @cizzymac @brym103 please re-test with today’s firmware update.

brym103, 2023-05-04:
> @cmcginty @cizzymac @brym103 please re-test with today’s firmware update.

Oh I sold my NT Mini Noir to get a MiSTer like 15 months ago at this point.  I wasn't going to wait around for an indeterminate amount of time for a bug fix when it had already been a year.  The release notes for 1.3 indicate that it was supposed to fix my issue though:

>Fixed Vertical Green Lines in games using CHR-RAM (Castlevania, Legend of Zelda, Mega Man 1 & 2, Rygar, many others)

Hopefully it worked for those of you still in the Analogue ecosystem.

cizzymac, 2023-05-04:
I can confirm the issue with vertical dotted lines on my flash cart and official Metroid cart are now FIXED with the new official firmware! Hopefully we get an updated JB firmware soon, so this issue can be put to bed once and for all!

closed 2023-05-04 by tjanas