SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #143: 'Bouncing' status bar on both Flintstones NES games

mlc4050, 2021-06-16:
Present in the official v1.2 firmware and 6.6 jailbreak firmware.  Tested on both physical carts and iNES 1.0 Headered roms.

tjanas, 2021-06-19:
Have you tried a rom with a correct NES2.0 header (via https://github.com/Kitrinx/NES_Header_Repair)? What do you mean by physical cart? Original NES cartridge of that game?

mlc4050, 2021-06-20:
Correct, the original NES cartridges

imjustcalledtwobrothers, 2021-06-22:
This issue is present in lots of games. The educational Mickey Mouse games also have it, aside from all the ones already reported. It might be good to make an overall bug for what seems to be one shared issue, rather than a separate bug for every affected game.

tjanas, 2021-06-22:
These issues should be brought up directly to Analogue as well, possibly via Twitter since there are outstanding issues with real original cartridges on official firmware. @imjustcalledtwobrothers & @mlc4050 could either/both of you tweet to them and share the link to it here?

mlc4050, 2021-06-23:
The bouncing status bar issue is present on the Vice Project Doom cart as well, with official v1.2 firmware, not present with v1.0 firmware.   Do you guys know of other status bar issues that should be brought up to them?

mlc4050, 2021-06-23:
https://twitter.com/MichaelRossow/status/1407841585832804354?ref_src=twsrc%5Etfw%7Ctwcamp%5Etweetembed%7Ctwterm%5E1407841585832804354%7Ctwgr%5E%7Ctwcon%5Es1_c10&ref_url=https%3A%2F%2Fpublish.twitter.com%2F%3Fquery%3Dhttps3A2F2Ftwitter.com2FMichaelRossow2Fstatus2F1407841585832804354widget%3DTweet

mlc4050, 2021-06-23:
Apologies for the terseness in describing the issue. I've only seen the issue firsthand on those 3 games.

tjanas, 2023-05-04:
@mlc4050 please re-test with today’s firmware release.

closed 2023-05-05 by mlc4050