SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #144: Crisis Force - distorted graphics level 3

mlc4050, 2021-06-16:
Distorted graphics  on level 3 of Crisis Force, rendering the game pretty much unplayable.  Present in the official v1.2 firmware and 6.6 jailbreak firmware. Tested on a physical cart and iNES 1.0 headered rom.

tjanas, 2021-06-16:
Have you tried a rom with a correct NES2.0 header (via https://github.com/Kitrinx/NES_Header_Repair)

closed 2021-06-19 by mlc4050

tjanas, 2021-06-19:
@mlc4050 so was your problem bad rom headers?

mlc4050, 2021-06-20:
When I reported the issue, I thought the NT Mini Noir was running the jailbreak 6.6 firmware.  I later found (after reporting the issue) that the console had been been backdated to the official v1.0 firmware, so I closed the issue. I didn't reconfirm if the issue is still present in the official 1.2 or jailbreak 6.6 firmware.