SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #145: Little Nemo freezing (cartridge) Jailbreak 6.6

zoogderrick2, 2021-07-07:
Little Nemo freezes during initial cut scene. If I skip the cut scene It also just after loading first level when running from cartridge. However when I dump the rom from the same cartridge using CopyNES it runs fine from SD card.

tjanas, 2021-07-07:
> Little Nemo freezes during initial cut scene. If I skip the cut scene It also just after loading first level when running from cartridge. However when I dump the rom from the same cartridge using CopyNES it runs fine from SD card.

Since the real cartridge has issues, have you reached out to Analogue via Twitter and email?

tjanas, 2023-05-04:
@zoogderrick2 can you please re-test with the latest firmware released today?

acidrain97, 2023-05-07:
Got through the entire sequence up to the title screen without issue. Using an ED pro though.

closed 2023-06-25 by tjanas