SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #148: Found issues with 2 games on most recent JB firmware as well as official firmware

handheldgamefan, 2021-08-05:
In a nutshell Mighty Final Fight crashes on title or soon after, but if i insert it into the famicom slot via an adapter to play US games it works perfectly fine. The second  game is Metroid (US NES version), haven't tested the FDS version. Anyway there are lines going down the screen like the cartridge is dirty or not seated correctly when I have taken apart and fully cleaned the game. I also confirmed it by rolling back to the stock official firmware 1.0 and both games work fine as they should. I dunno if these games were  already mentioned having issues, but just in case not I wanted to post here. I also e-mailed Analogue support about the official firmware already and was told the issue was forwarded to the team.

huhx0015, 2021-08-12:
I am also seeing the same vertical line issue with Metroid (US) NES version with the latest JB firmware.

BobsYurUncle, 2021-09-07:
I'm pretty sure the vertical line issue occurs only when playing off of cartridge.

R.C. Pro Am has a vertical line issue for me when using the cartridge. Loading a ROM off the SD card doesn't have any issue. However, reverting back to JB Firmware 6.5 got rid of the lines when playing off cartridge.

tjanas, 2021-09-07:
Has anyone heard back from Analogue Support regarding this issue?

handheldgamefan, 2021-09-07:
No sadly

Stopper33, 2021-09-18:
Definitely an issue with the Metroid cart. I've tried it with two copies. The copynes rom works fine.

PizzaCube, 2021-11-11:
the $500 console should get top priority for fixes .   its the most expensive one they sell.

also still waiting for the analog duo to show up since it says coming in 2021. (i doubt it)

W4JEW, 2021-11-11:
Do these games work well with the stock firmware but not with the jailbreak firmware?

Have you logged support cases with them directly (I.e. not here on GitHub for firmware they’re not responsible for)? If not, you should so they can track it. 

Analogue isn’t a very big company. There are only handful of employees - and Kevtris who’s effectively  is 50% of Analogue (as far as I’m concerned).

I’d venture to say they’re tapped out with trying to get Pocket and Analogue OS out the door and probably don’t have cycles to focus on these bugs.

They’re good about fixing issues. But you have to manage your expectations reasonably.

PizzaCube, 2021-11-11:
for a small team They seem to launch  a new product all the time and they just announced  analog OS before making sure the old products they sell are really stable...

the super nt and mega sg are getting all the attention

tjanas, 2021-11-11:
They said the existing consoles will not be getting the new OS, and I fear Analogue will forget about them and will abandon future firmware updates for them, just like the did with the Nt Mini that was released in 2017 (they never released any bug fix firmwares for it after 2017).

W4JEW, 2021-11-11:
AFAIK, the OS is not and was not ever intended for the other consoles. It's the foundation for the Pocket. There was going to be an OS on the Pocket in the first place because they started demoing the fact that you could use it for music creation. I think Analogue OS is more marketing than anything else.

The Pocket has been a work-in-progress for probably about 2 years at this point. They announced Duo but you'll notice they haven't started to take orders for them yet. And once they do start taking orders, if history repeats itself, it will be another year before it comes out. The SuperNt and MegaSg were the same way. They might do a little bit of work in parallel, but most is serial - one system after the other. Then a few firmware updates while they're working out the kinks.

But going back to the reason this post is on GitHub to begin with.... You mentioned the issue exists in the official firmware. Have you contacted Analogue to let them know? Have they at least acknowledged it and provided you with a support ticket number? If you haven't done these things and haven't received a support ticket number, I can tell you they probably don't know about the issue and they aren't tracking it to fix it.

tjanas, 2021-11-11:
I recommend everyone with an issue with real carts to bring it up with analogue, both in the form of submitting a ticket with them, as well as on social media (Twitter).

PizzaCube, 2021-11-14:
> They said the existing consoles will not be getting the new OS, and I fear Analogue will forget about them and will abandon future firmware updates for them, just like the did with the Nt Mini that was released in 2017 (they never released any bug fix firmwares for it after 2017).

i submitted a ticket with the jumpy status bar issues.... we will see if they do something . i dont know why they dont just open a github page and just read what people are saying.

scottlinux, 2021-12-19:
My I present a $500 NES console. Original cart.

[image]
[image]

handheldgamefan, 2021-12-19:
mine too

whitezombie2000, 2022-02-13:
I can confirm that Metroid and R.C Pro-Am have those lines on screen with real cartridges using JB 6.6 and the official 1.2 firmware. I dumped Metroid and the ROM ran fine in the 6.6 JB and on an emulator. I dumped R.C Pro-Am with MMC1 and I could not get the ROM to work on the 6.6 JB or on an emulator. Grabbed one from a ROM site and it loaded fine on the 6.6 JB. It might not be dumping the R.C Pro-Am ROM correctly too.

I can confirm that Mighty Final Fight crashes as described above using a real cartridge with JB 6.6 and the official 1.2 firmware. Oddly if I plug it into a Game Genie and launch it that way it will work fine. I could dump the ROM and run it just fine on the 6.6 JB and on a emulator.

Violator-80, 2022-02-19:
I am running the latest JB firmware (6.6) and I played my original cart of Rad Racer. I saw no vertical lines anywhere.

whitezombie2000, 2022-02-19:
Take a picture of the board and post it. In another post some people have the vertical lines with certain Castlevania, Rygar, Zelda revisions. Bottom line is the 1.2\6.6 JB has many problems with many real cartridges. It's amazing that Analogue isn't addressing these issues for a $500 console.

tjanas, 2022-02-19:
@whitezombie2000 in the meantime can you share photos of your Rad Racer pcb? @Violator-80 can you do the same?

whitezombie2000, 2022-02-19:
Here's a PCB photo of my Rad Racer cart.

[image]

Violator-80, 2022-02-19:
[image]

whitezombie2000, 2022-02-19:
It doesn't do this on 1.1. Many other people who have this issue on 1.2\6.6 JB have to revert to 1.1 to resolve this issue. As far as the power supply, yes I am using the Merryking PSU that came with it. I found another one with a little more weight that came with a Seagate external hard drive, barrel was the same size, specs matched the Triad adapter recommended by some from Digikey , 12V 1.5A , center +, outer -, and it does the same thing. So it's not an issue with the PSU.

[image]

ManeFunction, 2022-10-20:
Can confirm the same vertical bars issue on original Duck Tales, Top Gun and Guardian Legend. Broken on the latest 1.2, but working well on 1.1, both stock version and jailbrake. Wrote directly to Analogue support, highlighting this issue and one one, that 1.2 specific and doesn't exist on 1.1: Unplayable Broke Studio games, like Micro Mages, Project Blue and Mawthorne. Maybe others too, can't confirm.
[image]
[image]
[image]

whitezombie2000, 2022-10-20:
I think I've given up on Analogue fixing these problems. It doesn't seem like they have any interest in fixing these issues. I'm thinking about selling my Noir.

ManeFunction, 2022-10-21:
It's actually kinda works on 1.1 for me, but yes, it's a shame for a $500 product :(

scottlinux, 2022-11-04:
Contacting Analogue has no effect to anything getting fixed, sadly. This expensive product is abandoned and not supported which is unfortunate.

whitezombie2000, 2022-12-26:
The best review of the Analogue NT Mini Noir I have ever seen.

https://www.youtube.com/watch?v=jy-SgriEwC0

dataguardsec, 2023-01-02:
Add MegaMan original cartridge (1987 NES) to the list. Vertical lines. Downgraded to 1.1 and issues subside. I've emailed Analogue support.

SegaSnatcher, 2023-03-20:
> The best review of the Analogue NT Mini Noir I have ever seen.
> 
> https://www.youtube.com/watch?v=jy-SgriEwC0

Seems in bad taste to post a negative review video of the console in a github issues thread.   

This is not really the place for that.

whitezombie2000, 2023-05-04:
They released a new firmware today. I will test once I have the chance. Here are the release notes. How was it identified if a game uses CHR-RAM?

Fixed Vertical Green Lines in games using CHR-RAM (Castlevania, Legend of Zelda, Mega Man 1 & 2, Rygar, many others)

tjanas, 2023-05-04:
@handheldgamefan @huhx0015 @BobsYurUncle @Stopper33 @scottlinux @ManeFunction @dataguardsec please re-test your cartridges with the latest official firmware and report back https://www.analogue.co/support/nt-mini-noir/firmware/1.3

handheldgamefan, 2023-05-04:
> @handheldgamefan @huhx0015 @BobsYurUncle @Stopper33 @scottlinux @ManeFunction @dataguardsec please re-test your cartridges with the latest official firmware and report back https://www.analogue.co/support/nt-mini-noir/firmware/1.3

yes i can confirm the games I mentioned now work with no issues, really surprised they actually updated this very happy, I can also confirm Famicon Disk System games load fine now, I found that issue way after I made this post. So does this mean a new JB version is coming cause I'd love to play cartridge free lol

whitezombie2000, 2023-05-05:
I can confirm that Mighty Final Fight no longer crashes on v1.3 firmware. I can confirm that R.C. Pro AM, Rad Racer, and Metroid no longer have the vertical bars on v1.3 firmware.

BobsYurUncle, 2023-05-05:
tested R.C. Pro AM, Metroid, Ducktales and Guardian Legend. No vertical lines on 1.3

scottlinux, 2023-05-05:
There is a new firmware update from analogue, haven't test yet but wanted to give a heads up: 

https://twitter.com/analogue/status/1654144815226064896

closed 2023-05-05 by tjanas

scottlinux, 2023-05-05:
Can confirm no vertical lines on Metroid original cart, and also Battletoads no longer freezes after the second stage. Thank you!!

whitezombie2000, 2023-05-05:
@scottlinux or anyone By any chance do have the game Dragon Spirit? Someone reported that it didn't work despite contacting Analogue. They claimed it didn't work on the new firmware. They didn't give specific details on the issue but I reached out to the person to investigate.

handheldgamefan, 2023-05-05:
[image]
It seems to be fine with my japanese version