SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #155: Dragon Warrior III and IV can't rip in COPYNES (JB 6.6)

FremontDZ, 2022-01-18:
This is a known issue but I didn't see it listed here in the issues so I wanted to get it on the radar. 

I've tried editing headers, triple-checking the mapper, and still can't get either game to rip correctly. DW 3 doesn't load at all (black screen) and DW 4 loads for a second, then goes all to bits (artifacts in the graphics). 

Carts play fine.

PizzaCube, 2022-01-22:
startropics does not rip either

Cheesemeister3k, 2022-11-15:
[SXROM.zip](https://github.com/SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak/files/10013441/SXROM.zip)
For DW3&4, use SXROM.BIN from this ZIP file adapted from nescartdb.com/plugins.

For ST, use HKROM.BIN.

FremontDZ, 2022-11-21:
Thanks! I'll give it a try!

FremontDZ, 2023-01-17:
@Cheesemeister3k worked like a charm, thanks so much for sharing! Glad to have "saved" my Dragon Warrior 3 characters as well as running the game off the SD card vs. the cart. Closing the issue.

mariomadproductions, 2023-01-17:
@FremontDZ i think this issue should probably be re-opened, so other people can see the solution

FremontDZ, 2023-01-17:
Done!

tjanas, 2023-01-18:
Issue should remain open until the fixes are incorporated in JB v6.7 or later.

FremontDZ, 2023-01-29:
@Cheesemeister3k FYI - the SXROM you provided didn't actually work for Dragon Warrior 4 (it worked great for 3). In 4, it copies the game but doesn't recognize the saves. Then, when you start, the text is all gibbrish.

Cheesemeister3k, 2023-01-30:
@FremontDZ I tried the DW4 save file functionality and it appeared to work correctly when the battery checkbox was checked.

tjanas, 2023-06-26:
@FremontDZ @PizzaCube please create an issues at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB if the issue still exists with JB v6.7.

closed 2023-06-26 by tjanas