SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #156: "Unsafe" (Speedrunners and the need for a 4th video option)

charlesmadere, 2022-04-13:
Hi everyone,

I think most of us are aware of the Analogue consoles 3 main video settings: **Fully Buffered**, **Single Buffer**, and **Zero Delay**. Each are within the standard featureset of the Super Nt, Mega Sg, and Nt mini Noir. All of these are perfect for HDMI and work great for their use cases, but then also inevitably come down to a compromise.

With **Fully Buffered**, the Analogue console will run at the exact timing as the original console, but with additional input lag in order to meet the HDMI’s standard 60Hz spec.

With **Single Buffer**, the Analogue console will run at the exact timing as the original console, but with variable input lag in order to meet the HDMI’s standard 60Hz spec.

With **Zero Delay**, the Analogue console will either overclock or underclock to meet the HDMI’s standard 60Hz spec. Therefore it has subtly gone out of spec with the original console, but with the bonus of absolutely no input lag.

One could argue, and I think it would be valid, that the above should cover just about any person’s use case. That may be true… but speedrunners need better. Especially for people pushing for top times. Inevitably there are massive compromises with each of the above options.

I speedrun Super Metroid any% on my Super Nt and have a PB of 42:37, a time that puts me at #15 in a game with over 1,000 submitted runs on speedrun.com. And I did this on Zero Delay. Considering that I lose about 1 second every 10 minutes under this setting, does that mean my PB should be about 42:33? Or perhaps I should run on Single Buffer, because then that way I don’t lose time? But that sounds terrible right, to play at high level with variable input lag? So then, maybe I should play at Fully Buffered, to not lose any time at all, and also not have variable lag… but then I’m accepting having even more input lag?

No, this is not why I or any of my friends bought Analogue consoles. We didn’t buy them to have to make compromises like this.

Additionally, the Super Nt and Nt mini Noir are actually advantaged over Mega Sg in this conversation. The Mega Sg actually has to overclock in order for Zero Delay to hit 60Hz, whereas the Super Nt and Noir both downclock. This means that the Mega Sg on Zero Delay should probably be literally banned by any serious speedrun leaderboard.

I would like to put a call out for a 4th video option: “Unsafe”. This would output the console’s true native Hz, HDMI spec be damned, over the HDMI connection. With this option, there is no compromise.

Clearly this introduces a new problem though: what can Analogue do to prevent people from selecting this option, while using hardware that doesn’t support the weird Hz, and then suddenly they have a blank picture on their screen? How does someone recover from this?

I suggest the following solution:

* User navigates to the console’s video modes
* User selects "Unsafe"
* A dialog pops up and says "Warning, your TV may not be able to handle this input. Super Nt will switch to the native console framerate, and then ask you to enter a specific series of button prompts. In 15s if the button prompts have not been entered, Super Nt will fallback to the current video option."
* User confirms
* User then waits for their TV to switch to the different framerate (if it works)
* User then enters the button combo, and they're now good to go with the native Hz
* Alternatively, the user can't see anything on their TV, and after 15s, the Super Nt reverts to the previous video option

\^ The above solution was sent by me to Analogue in December 2021, which a response shortly after:

>Hi Charles, I have passed along your request to our development team. Thank you very much and have a good day! Best Regards, Analogue Support

Now I appreciate this email, it looks great. But given the dearth of general firmware updates, and how freaking strongly I feel about this issue, I am actually seriously considering buying an SNES Jr, getting it RGB modded, buying HD Retrovision cables, and a Retrotink5x, all for a measly \~4s of time save. And you know what, this solution, even though it would work, sucks because (A) it goes against the “perfect” aspect of Analogue consoles (B) it is very expensive (C) a Retrotink5x actually introduces an additional 4ms of input lag.

Thanks so much for your time, please spread this issue around and get the word out if you can.

smCharles

For the nitty gritty on this topic, the following site has all the old consoles and framerates nicely listed and detailed: [https://emulation.gametechwiki.com/index.php/Resolution](https://emulation.gametechwiki.com/index.php/Resolution)

Fellow Super Metroid runners and friends of mine who would also love having this option available to them:

* Imyt
* Eddie
* FreyasSpirit
* CynanMachae

Further conversation on Reddit about this issue: https://www.reddit.com/r/AnalogueInc/comments/u33x4s/unsafe_speedrunners_and_the_need_for_a_4th_video/

eltees, 2022-04-15:
There could actually be one other (potential) benefit for some people, at least i've found something similar using other hardware which has allowed that output over hdmi to some of my older tv's, and that would be reduced display input lag.  I have two very nice looking but aging lcd screens that both unfortunately normally give about +4 frames of input lag over crt (doing side by side photo comparsions).  This is true with pretty much any native hdmi source and since both of these tv's are 2008-2011 vintage, that is all too common.

Weirdly though when i tried using other hardware with support for 'off brand' timings to both of these older tv's (2008 sharp and 2011 panasonic) not only did they both did synch up and work, but more strangely, they both went from 4 frames of display input lag to 2 frames of input lag (Again direct captured vs a crt splitting the same signal and frame timing test roms).  I think potentially the 'off' timings broke/disengaged some of their 10+ year old display buffering or internal processing, forcing them to just display as fast/raw as they could, and they went from unplayably slow to actually 'acceptable' (if not ideal).  I know it's kind of a corner case of a corner case but just wanted to throw the extra two cents in there.

tjanas, 2023-06-26:
@charlesmadere if this feature request is still wanted, please open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB

closed 2023-06-26 by tjanas