SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #158: Miracle Piano teaching system is unplayable

imjustcalledtwobrothers, 2022-07-11:
The game requires the piano to be plugged into slot 1 and the controller in slot 2 to control the menu. But you can't make any selections with the controller on the NT Mini Noir, so the game is unplayable. 

I am running from a rom because I can't find my cartridge (which I really do own, I mean, I have the piano why would I steal the $20 cartridge?). It'd be good if someone else (if they exist) could test on original firmware or with the original cartridge. If mine turns up I will report back on this.

tjanas, 2023-05-06:
@imjustcalledtwobrothers have you found your Miracle Piano cartridge yet?

tjanas, 2023-06-26:
@imjustcalledtwobrothers please re-test with JB6.7 and open an issue at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB if the issues still persists.

closed 2023-06-26 by tjanas