SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #160: G.I. Joe real cartdrige works on 1.0 Original firmware but not on Jailbreak

TT-Hipster1941, 2022-08-27:
Hello SmokeMonster,

I have a GI Joe : A Real American Hero NES cartdrige that works perfectly on my original console but not on my Analogue Nt Mini Noir V2. I've been running a couple tests to figure wich version is working or not

- Firmware 1.0 from Analogue works
- Firmware 1.2 from Analogue does not works
- Jailbreak Firmware 6.2 works
- Jailbreak Firmware 6.5 works with sound issues
- Jailbreak Firmware 6.6 does not works

All the tests have been done without moving the cartdrige to exclude false positive. Cartdrige works 5 / 5 attemps on my OG NES. *I know it's a rare thing lol*

tjanas, 2022-08-28:
Since the issue also exists with official Analogue firmware 1.2 (the latest firmware as of this post), could you contact Analogue Support and report back with their response? Maybe they will have a fix soon?

TT-Hipster1941, 2022-08-28:
Email has been sent to support, will come back with answer. Thanks for the help.

TT-Hipster1941, 2022-08-29:
Response from Analogue : "Thanks for the information. We have passed along the information to our development team. This will be addressed in future updates. In the meantime, please install the firmware v1.1 to avoid these issues."

NB : I haven't tested 1.1 yet and did'nt mentionned it in my email.

whitezombie2000, 2023-05-05:
@TT-Hipster1941 I see you reported on Twitter that this game is fixed with v1.3 and that's great news! This issue should be closed now.

closed 2023-05-05 by tjanas