SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #163: Palette selection is ignored when using analog video output

Bohman-Timothy, 2022-12-31:
When the analog video output is connected via a compatible VGA adapter (purchased separately from a third party) to a compatible TV, the selected color palette option is ignored by the Analogue Nt Mini Noir. When choosing a palette in the menu, I can see a preview of the intended color palette on the game screen in the background, but as soon as the menu is closed, the colors revert to a default palette.

The simplest way to make the effect of the bug obvious is to load an NES/Famicom game (cartridge or ROM) and enter the Nt Mini Noir's system menu at any point in the game that will provide a multi-colored background image while in the system menu (pausing may be helpful, but various games' title screen can also work), select the Black & White palette, observe the greyscale colors in the game in the background, then exit the menu (i.e. return to the game). The game screen will have its colors restored. Entering the system menu again will show the selected Black & White palette is active again in the game screen behind the menu.

It doesn't matter which palette I select, whether a built-in palette option or a custom palette I have on my SD card—when the system menu is closed, the colors revert to a default palette (that for some reason doesn't seem to exactly match any of the built-in palette options).

[End of bug report]

As a side note, I have to mention that outputting the video via an S-Video connection gives excellent picture quality on a CRT television, and in my opinion, NES games have never looked better (aside from currently not being able to show a palette I prefer). Of course, the "CRT filtering effect" is present, which is good for blending the harsh pixelated outlines of everything in NES games, yet the S-Video signal gives the image a clarity above anything the original NES console ever output. For anyone who has a preference for viewing NES games on a traditional CRT screen for the visual "filtering," and who has the inclination to get a crisp-yet-pleasantly-and-minimally-blended image on an actual CRT television (that has an S-Video input), I highly recommend getting a compatible analog video adapter for the Nt Mini Noir and an S-Video cable to appreciate the enhanced visuals for yourself. (For those interested, the adapter I got is the Monoprice VGA to S-Video/RCA (Composite) Adapter Cable, and it has worked well for converting to either an S-Video or a composite/RCA signal.)

tjanas, 2023-01-02:
I believe that is intentional; the colors of the analog video output via composite / s-video should replicate the NTSC video output from a genuine NES console. The colors would then depend on the CRT television.

Bohman-Timothy, 2023-01-03:
I don't know how the Analogue Nt Mini Noir generates the colors for analog output, but the colors don't match what my NTSC-region NES console outputs when I compare both systems' video over a composite connection on the same television (and when I compare the two systems' video colors on a second television). One color that stands out to me as being noticeably different is the primary yellow color (its position in the sequence of colors in the palette is represented in hexadecimal as 28), which, when output by the Analogue system, is more greenish—a slightly greenish yellow, compared to the yellow from the original NES system. The default NT Mini Noir analog palette is good, but it has several colors that I notice not matching the original NES system's palette, even though they are recognizably near the original system's actual colors.

Even if the Nt Mini Noir is emulating the method used by the original NES to output its colors, not all of these colors are the exact colors I've seen the actual NES console output on various televisions in my lifetime. I'd like an option to be added to use either the default palette / color-generating method or a custom palette, which could potentially get closer to the colors I see my original NES generate. I'd also like to be able to use custom palettes that don't represent the original colors, such as a palette I have that simulates 4-color CGA computer graphics of cyan, magenta, white, and black. I'd like to be able to try out on a CRT television different palettes that aim for accuracy, as well as novelty color palettes. So this can be considered a feature request if it isn't accepted as a bug report.

(The CGA palette can still allow some games to be playable, depending on the way a game uses contrasting colors. The palette works well for Double Dragon II - The Revenge from beginning to end, for example.)
[image]

Though, as I've thought more about it, maybe this is something that should be added to the official firmware. @tjanas What is your take on all this?

tjanas, 2023-01-03:
If there are significant differences in the NTSC video signal from the Noir, I am surprised that it wasn't noticed by any of the reviewers of the Nt Mini Noir.

tjanas, 2023-05-04:
@Bohman-Timothy can you please re-test with the latest firmware released today?

Bohman-Timothy, 2023-05-28:
I have compared the latest jailbreak firmware (version JB6.6) and the recently released official firmware (version 1.3), and I don't notice any change between the new and the old firmware regarding the colors used when outputting via an analog video signal (composite or s-video). I compared in a few games and in a public demo ROM that can display the entire NES palette via colored bars: "NES Color Bars Demo V2 by Quietust (PD)."

Side note: my go-to game for comparing the primary yellow color is Zelda II: The Adventure of Link, since you can quickly start a saved game and walk out of the palace to see the yellow roads, which are wide enough to see the base color without color-blending interference from the colors next to the path.

SamuraiSamH, 2023-07-18:
Can confirm, I have this issue too and it definitely bugs me. It would be so so great if I could actually get the colors I want when playing on my CRT, as the image otherwise looks beautiful in RGB.

tjanas, 2023-07-18:
@Bohman-Timothy @SamuraiSamH please re-create the ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB for visibility. In the meantime I will close this issue here, since new JB updates are now released thru the other repo.

closed 2023-07-18 by tjanas