SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #165: Crystalis screen shake when text box open on 6.6 (cart + rom)

nobleEightfold, 2023-03-08:
On firmware 6.6, the screen graphics shake when a text box is shown on screen, e.g. when talking to characters in town. Happens on real cart and on rom.

The issue does not happen on the NT mini when on firmware 6.5 when playing via cart or rom and does not happen on a real front loader NES playing from a retail cart.

For context, this issue also happens in some emulators and is as far as I understand related to the timing of the MMC3 A12 read/write snoops used for the scanline counter.

tjanas, 2023-05-05:
@nobleEightfold could you please re-test your cart with the latest official firmware released today?

nobleEightfold, 2023-05-05:
@tjanas Just tested on official cart on official firmware 1.3 - this issue is fixed and the text box in Crystalis behaves like it does on a front loading NES.

closed 2023-05-05 by tjanas