SmokeMonsterPacks/Nt-Mini-Noir-Jailbreak #166: NES core: faulty reads from WRAM locations $6000/$6001 on some TNROM boards

sircodalotkob, 2023-03-21:
Recently, we had the opportunity to give our currently developed NES game a test spin on a borrowed Analogue NT Mini Noir 2.

As the game had been thoroughly tested on real NES consoles, we were very surprised to see it behave completely weird on the NT.

Our game uses a TNROM board (MMC3 mapper) with WRAM at $6000. We found out that the NT fails to read back the previously written values from the first two WRAM addresses $6000 and $6001.

We could reproduce this with TNROM boards from three different suppliers as well as the original Everdrive N8. Also, it happens with all firmwares, including the official 1.0, 1.1, 1.2 and JB 6.2, 6.5, 6.6.

Failing to read from WRAM could lead to all kinds of strange issues, from corrupted savegame data, to glitches in the game world, to crashes. If there is an issue here, it might affect other games as well.

tjanas, 2023-03-22:
Thank you for the report. Could you try contacting Analogue to see what their customer support has to say?

tjanas, 2023-05-06:
@sircodalotkob What are your findings with the latest firmware (official 1.3)? What did Analogue say?

https://www.analogue.co/support/nt-mini-noir/firmware/1.3

sircodalotkob, 2023-05-26:
The device we experienced the issue with was a borrowed one, so we haven't been able to re-test with the latest official firmware (1.3) yet. However, there _may_ be a chance to do so in July. If the issue still persists, we're going to contact Analogue directly.

tjanas, 2023-05-27:
Kevtris on [Classic Gaming discord](https://discord.gg/UDu5ztY) might have some insight?

tjanas, 2023-06-26:
@sircodalotkob if you are able to re-test with the latest official firmware or JB6.7, and the issue still persists, please open a ticket at https://github.com/analoguejb/Analogue-Nt-Mini-Noir-JB. In the meantime I will close this issue. Further issues should be created that that new repo.

closed 2023-06-26 by tjanas

sircodalotkob, 2023-06-26:
We'll be giving it a try on 1st of July and report back on the other repo.